dhalperi/linux-80211n-csitool-supplementary #153: Can I measure CSI on an external NIC?

soltanaghaei, 2016-02-17:
Hi, I want to use the CSI tool for my project, but I would like to run it on my own laptop. As my laptop does not have an Intel 5300 NIC, I'm thinking of connecting the required network card externally using a Mini PCI-E Card to USB 2.0 Adapter Card. Can you please guide me if the driver modifications and CSI measurements work in this case? 

Thank you.

soltanaghaei, 2016-02-17:
Update: If using Mini PCI to USB doesn't work, what configuration do you suggest to get the CSI results easier and faster without further issues?

dpward, 2016-02-17:
I'm pretty confident that would not work.

According to (http://electronics.stackexchange.com/questions/26961/wifi-module-says-mini-pci-e-format-with-usb-host-interface-what-does-this)[this], there are some Wi-Fi adapters that physically have a Mini-PCIe form factor and fit into a Mini-PCIe slot, but they are really USB 2.0 devices. This is apparently because the Mini-PCIe card edge connector can provide a USB 2.0 interface by using certain pins. The type of adapter you are describing would presumably just re-route those pins onto a physical USB 2.0 port.

The Intel Wi-Fi Link 5300 is truly a PCI Express device however, not a USB device. Does your laptop have a Mini-PCIe slot (perhaps occupied by an existing Wi-Fi adapter that you could take out for the time being)? If not, you need to use another system that has a real Mini-PCIe slot.

soltanaghaei, 2016-02-18:
First of all, I would like to thank you for providing such an amazing
community for using this helpful tool in research projects and Thank you
very much for your guidance in my problem. Actually, my idea is to find out
a way to use CSI tool in a small portable device such as raspberry pie. I
found some hardwares that support mini-PCI, but I couldn't make sure if the
tool will work on those devices or not. I would be grateful if you can take
a look on these options and guide me in this regard. Based on the
explanations in CSI-tool website, I realized that the tool requires an
intel system with specifically intel wifi 5300 NIC. intel Galileo
<https://www.cooking-hacks.com/documentation/tutorials/intel-galileo-tutorial-using-arduino-and-raspberry-pi-shields-modules-boards/>
is an intel microcontroller with mini-pci slot. What's you idea about this?
Is there any other assumption in the tool which may make it impossible to
use on other platforms? The other idea is if we can extract CSI values on a
small software defined radio such as BladeRF
<https://www.kickstarter.com/projects/1085541682/bladerf-usb-30-software-defined-radio>
or any other type of software radios. How much do you think it's possible
to implement a similar tool or modify this tool for software radios?

I would be grateful if you can share with me other possible solutions to
run CSI tool in a small portable microcontroller or semi-computer.

Regards

egaebel, 2016-02-29:
You need to run this on a device that supports Linux. Microcontrollers like the Intel Galileo and Arduino are not going to be able to run software as complex as this. 

This microcontroller has a built in pci-e slot and can run linux. This is probably your best bet. 
http://microcontrollershop.com/product_info.php?products_id=7057

If it works I'd love to know about it.

egaebel, 2016-02-29:
As for running on your own laptop, I was lucky enough to have an express card slot, which supports pci-e and I was able to find an adapter to pcie for it. If your laptop doesn't have such a slot, I'd recommend looking on Ebay for a cheap one.

skocharllge, 2017-04-24:
@egaebel 
I just got a confirmation email from GoEmbedded representative(Please see below) that 
http://microcontrollershop.com/product_info.php?products_id=7057
does not support PCIe signals .Is there option people tried where in CSItool was ported to some other hardware apart from laptop and desktop.

Hi Srinivas,

 

Yes, you cann’t use Intel5300board WIFI chip with beagel bone or raspberry PI boards. The beaglebone and raspberry Pi don’t support PCIe signals.

Our B2A carrier board is use miniPCIe socket. The socket have 52pins, it can support PCIe signals and USB signals. We only use USB signals.

I attached the B2 SCH for your reference.

 

Best regards,

Yukun

 

From: srinivas [mailto:s.kocharlakota@lge.com]
Sent: 2017-04-24 10:39
To: GOEMBED; support@goembed.com
Cc: 'Wuxiuxin'; Joon Lee/LGEUS Integrated Living Home Innovation Team(joon1.lee@lge.com); Viswanath Ganapathy
Subject: Re: Regarding ssh password SBC335x-B2A

 

Hi Yukun

Small clarification on the board we bought is listed here

http://microcontrollershop.com/product_info.php?products_id=7057

It has MiniPCIe interface as shown in the figure

BR
Srinivas

Laurent-St, 2018-09-14:
Hi everyone,

In the end did someone managed to get the tool working with an external wifi card using a PCIe-to-USB adapter?

Missilazi, 2020-04-27:
Hello all,

I am trying to achieve the same thing and connect an intel 5300 NIC to a raspberry pi like device. 

I've found this device that seems to support PCIe.
https://www.antratek.com/nanopi-neo4

Will this work?

Thank you,
Marcello

skocharllge, 2020-04-30:
I do not see a reason why it should not work