dhalperi/linux-80211n-csitool-supplementary #154: Injection and Monitor at the Same Time?

egaebel, 2016-02-29:
Hi, I've been able to get two laptops to communicate using injection and monitor, but I'd like to know how to allow both laptops to monitor and inject at the same tame so that the two can communicate back in forth in some sort of usual manner.

Is it necessary to set up another virtual interface?

Thanks!

tylervv, 2016-02-29:
I've been trying to get the two laptops to communicate using injection and monitor but I'm not receiving any packets on the receiving side.  Any advice?  I could send you the steps I'm doing.

egaebel, 2016-03-01:
I found that I had to modify the injection and monitor scripts slightly. Here's what I am using

injection:
```
#!/usr/bin/sudo /bin/bash
WLAN_INTERFACE=$3
SLEEP_TIME=1
echo "Bringing down $WLAN_INTERFACE"
ifconfig $WLAN_INTERFACE down
sleep $SLEEP_TIME
echo "Deleting mon0.........."
iw mon0 del
sleep $SLEEP_TIME
echo "Bringing up firmware...."
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi debug=0x40000
sleep $SLEEP_TIME
echo "Running ifconfig on $WLAN_INTERFACE, looping until success...."
ifconfig $WLAN_INTERFACE 2>/dev/null 1>/dev/null
sleep $SLEEP_TIME
while [ $? -ne 0 ]
do
    ifconfig $WLAN_INTERFACE 2>/dev/null 1>/dev/null
    sleep $SLEEP_TIME
done
echo "Adding monitor to $WLAN_INTERFACE...."
iw dev $WLAN_INTERFACE interface add mon0 type monitor
sleep $SLEEP_TIME
echo "Bringing mon0 up...................."
ifconfig mon0 up
sleep $SLEEP_TIME
echo "Killing the notorious wpa_supplicant....."
killall wpa_supplicant
sleep $SLEEP_TIME
echo "Setting channel on mon0 to $1 $2 ........."
iw mon0 set channel $1 $2
```

monitor:
```
#!/usr/bin/sudo /bin/bash
SLEEP_TIME=1
WLAN_INTERFACE=$3
echo "Bringing $WLAN_INTERFACE down....."
ifconfig $WLAN_INTERFACE down
sleep $SLEEP_TIME
echo "Loading up firmware......"
modprobe -r iwlwifi mac80211 cfg80211
#modprobe iwlwifi connector_log=0x1
modprobe iwlwifi connector_log=0x5
echo "Setting $WLAN_INTERFACE into monitor mode....Looping until success..."
# Setup monitor mode, loop until it works
iwconfig $WLAN_INTERFACE mode monitor 2>/dev/null 1>/dev/null
while [ $? -ne 0 ]
do
    iwconfig $WLAN_INTERFACE mode monitor 2>/dev/null 1>/dev/null
done
sleep $SLEEP_TIME
echo "Bringing $WLAN_INTERFACE up......."
ifconfig $WLAN_INTERFACE up
sleep $SLEEP_TIME
echo "Killing the notorious wpa_supplicant"
killall wpa_supplicant
sleep $SLEEP_TIME
echo "Setting channel to monitor on $WLAN_INTERFACE to $1 $2" 
iw $WLAN_INTERFACE set channel $1 $2

```
The sleeps probably aren't all necessary.

And I also do the non-script steps specified at: 
https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection

egaebel, 2016-03-17:
I got something functionally equivalent working...I'll post it here later.

closed 2016-03-17 by egaebel

tylervv, 2016-04-05:
Hey @egaebel ,

Could you help me on getting both to inject and monitor at the same time?  I finally got them both to be able to send and receive separately.  Thank you!