dhalperi/linux-80211n-csitool-supplementary #155: The problem about Monitor mode

shaozhuchuang, 2016-03-07:
hello, I have get csi data in monitor according to https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection, but I don't know why the transmitter and receiver need connect to one AP? Besides, If the AP have two antennas, the csi data will be 2*3*30, why first dimension is 2. Are there any relations between csi data and AP? What are the relations between csi data and transmitter or receiver??Thank you!!

asifhanif1122, 2016-03-07:
As far as my understanding is concerned, we cannot distinguish data of more than one transmitting antennas at receiver side using sir Daniel's modified drivers. But to be honest you have to confirm from  sir or his colleague. You can get same data on multiple receiving antennas. If you are using two laptops (one as transmitter and other as receiver), you don't need to connect to AP to get CSI data. Now coming to your question about relation among CSI, transmitting antennas and receiving antennas ; the CSI data we receive have dimensions 1x3x30 with condition that your all three receiving antennas are connected.

shaozhuchuang, 2016-03-08:
@asifhanif1122 You said you don't need to connect to AP and can get csi data if you use two laptops, why not use desktop? If the transitter and receiver don't connect to an AP, i can't get csi data in receiver. Can you explain it in detail? Thank you!

asifhanif1122, 2016-03-14:
Dear @shaozhuchuang can you please give glimpse of you network scenario. Actually i was using one laptop 's NIC as transmitter whose purpose was to transmit CSI packets and i set NIC of another laptop in monitor mode. Receiver NIC was silently listening and receiving CSI packets. In this case i didn't need to have an AP to connect with.

simondchen, 2016-04-09:
Dear @asifhanif1122  I am using monitor mode, and I follow the https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection, but I can't get any csi data. I can get csi data in the AP mode, but the AP mode is not stable, so I change to monitor mode, but no csi data captured. I'm using ubuntu 11.04, is this version too old

simondchen, 2016-04-09:
@asifhanif1122  solved! thanks

asifhanif1122, 2016-04-10:
thats great @simondchen

Zhijing1128, 2017-01-27:
@dpward @dhalperi @asifhanif1122 "Monitor mode: The packet is addressed both to and from a hardcoded, fixed address 00:16:ea:12:34:56. " Why it's a hardcoded, fixed address? So I cannot do normal sniffer of other devices with different mac address?

88wx, 2017-11-14:
@simondchen I have the same question as you, I can get csi data in the AP mode, but I can't get any csi data in monitor mode. Could you tell me how to solve it? Many  thanks! 
Maybe there are some methods I can test this is  the problem of transmitter or receiver? I need your help. Hope your apply, thank you! @asifhanif1122 @dpward