dhalperi/linux-80211n-csitool-supplementary #157: about the phase of the subcarries

ozawali, 2016-03-28:
@dpward @dhalperi  I find that why the phase of 15th and 16th subcarry like this
[image]

asifhanif1122, 2016-03-28:
@ozawali  I want to ask question about stability of CSI though it is not related to your issue. Do you get stable CSI whenever you perform experiment. I am facing trouble in this regard. Whenever i do experiment i received CSI that changes with time despite the fact that environment is static. I am using 5300 NIC on tx and rx. Do you know possible factors for this change in CSI data?

ozawali, 2016-03-28:
Yes it's stable

asifhanif1122, 2016-03-28:
Can you suggest me possible steps to be taken to make sure that every time stable CSI is received. I am using get_scaled_csi function to get CSI from .dat file in Matlab. 
I have attached following picture to show that my CSI is not stable along packet index axis.

[image]

I am using monitor mode to receive packets ; transmitter and receiver both have 5300 NIC.

ozawali, 2016-03-28:
the attenuations is uselss

asifhanif1122, 2016-03-28:
How are you unwrapping the phase of CSI? Can you tell me technique or useful link so that I may leverage phase information of CSI.

cedivad, 2016-04-06:
I'm not sure if this answers your question, but I also saw that phase issue around the middle carriers in the majority of my tests so far, I believe we just have to live with it.

ozawali, 2016-04-06:
@cedivad  I think this issue must be the problem of the hardware,because according to the theory ,the middle carriers shouldn't be flat.so I hope some one know it @dpward @dhalperi

cedivad, 2016-04-10:
I believe it's because of the zero sub-carrier. I've just finished reading that the zero subcarrier (that is around where you see the interference in the graph, and not subcarrier zero in that axis) has no data associated with it, because it overlaps with the DC offset, or 0Hz.  

I really don't understand how that is possible, since that if you move your RF frequency to IF at center frequency 0Hz you would have every subcarrier < 15 in that graph at a negative frequency, that I don't think is a thing. 

But anyway, this is what Chronos' paper says under paragraph 4.

ozawali, 2016-04-11:
@cedivad @dpward @dhalperi  Yes  I have read that Chronos,the middle subcarry that is the 16th subcarry,it phase is simliar to the 15th that you see in the figure. I just can't know that the relation of phase between the 15th and the 16th

ccjack, 2016-04-25:
@ozawali Hi, Could you tell me how you unwrapped CSI phase? I am going to redo what you have and compare with the phase info from USRP.

libing64, 2016-05-25:
@ccjack You can try matlab built-in function 'unwrap'.

mazilaile, 2016-11-29:
@cedivad ,you mean that the 16th subcarry is the OFDM zero subcarrier,so we can not measure on zero-subcarriers?

mazilaile, 2016-11-30:
@ozawali  @libing64  Could you tell me why the phase of 15th and 16th subcarry  is so flat??  Thanks !

hot-dog, 2016-12-05:
@libing64 @ozawali @cedivad @asifhanif1122 I have a question that bothers me a lot, why the unwrap csi phase is nearly liner, can you tell me the reason or give me some materials related to it. Thanks!