dhalperi/linux-80211n-csitool-supplementary #158: Sampling Rate of CSI Data in Monitor Mode

asifhanif1122, 2016-03-28:
@dpward @dhalperi  I am confused about sampling rate of CSI values. On which factors does it depend? I am using 5300 NIC in both transmitter and receiver and my mode of receiving packet is monitor mode. 
If i send 2000 packets at data rate of 6.5Mbps and delay between two consecutive packets is 100 mili-second and length of each packet is 100 bits , what will be sampling rate of data?

ozawali, 2016-03-28:
@asifhanif1122  please see the FAQ

zainalabidenakhtar, 2016-06-07:
@asifhanif1122  how you set these sampling rates, data rates and delay between two consecutive packets? because i also want to do so that but could not done please help with commands, thanks

chendachong, 2016-09-07:
Would you tell how to set the sampling rates, thank you very much!

mikeWangJC, 2016-09-23:
@asifhanif1122 Is there any ideas about get the whole CSIs from 128 sub-carrier

asifhanif1122, 2016-09-24:
CSI is actually CFR. If you intend to get CIR, channel impulse response , you have to take IFFT of CFR.

88wx, 2018-02-05:
@asifhanif1122 @ozawali  @chendachong Would you tell how to set the sampling rates in monitor mode, thank you very much!