dhalperi/linux-80211n-csitool-supplementary #159: unloading drive

qincaixj, 2016-03-29:
to my best knowledge,I have taken care of all instructions well. I am facing an issue at step 5，when unloading drive.I input"sudo modprobe -r iwlwifi mac80211" to unload the driver.it showed "

tara@ubuntu:~$ sudo modprobe -r iwldvm iwlwifi mac80211
rmmod: ERROR: missing module name.
modprobe: FATAL: Error running remove command for iwlwifi

 I am stuck at this point. Any thoughts?Thanks a lot,I am looking forward to your reply.

zhangshaohu, 2016-04-24:
me either, have you solve this problem.

dhalperi, 2016-04-25:
Please search the web for how to list modules, load, remove kernel modules. You will need fluency with these low-level kernel operations in order to be able to navigate this tool.

My guesses: either you have not loaded the module correctly in a previous step (likely), you built the driver into the kernel instead of as a module (unlikely), or the driver module name as changed (least likely).

`lsmod` is probably a useful command.

dpward, 2016-05-25:
In step 5, you only need to run
```sudo modprobe -r iwldvm iwlwifi mac80211```
in the event that
```sudo modprobe -r iwlwifi mac80211```
returns an error (which depends on a certain configuration file in /etc/modprobe.d). I've reformatted the instructions to make this more obvious hopefully.

nafiul-nipu, 2017-05-27:
hello guys,
i am facing the same problem. can anyone tell me how you guys solved this problem?