dhalperi/linux-80211n-csitool-supplementary #160: how to send csi information to a server?

shaozhuchuang, 2016-03-30:
Now, the csi information is stored in a file, but I want to the receiver can send data to another computer when the receiver receive packages?  how i modifiy some code, if you have any idea, please tell me. Thank you!

egaebel, 2016-04-21:
I don't think this relates to the CSI tool. You can just send the file to a server in the usual way over the Internet. If you want to send the file over the local wireless channel using injection mode then look at the random_packets.c file here: https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection and alter the code to send packets from a file instead of randomly generated packets.

haimapi, 2016-04-28:
I used to set up a web server, apache, make the service on.
when collecting data, make the .dat files stored at the web 
directory, such as 
./log_to_file tmp.dat /var/www/
Then another host can access this .dat file