dhalperi/linux-80211n-csitool-supplementary #162: Sending and receiving simultaneously

tylervv, 2016-04-08:
I was able to get two computers to send and receive  separately using injection scripts and monitor mode but now I want both computers to send injections as well as receive at the same time. Any tips? Thank you!

egaebel, 2016-04-10:
You'll have to switch off between the two since the firmware needs to be set into a different mode to do each. So you'll have to transmit, switch over, receive, switch over, etc. I got this working pretty well.

simondchen, 2016-04-11:
Dear @egaebel, I encounter this problem as well. Can you explain in detail how to switch over, using a shell script? Can we switch over in a program?

simondchen, 2016-04-11:
@tylervv I succeed in sending and receiving simultaneously. Using wlan0 interface to send and receive at the same time without adding a mon0 interface.

egaebel, 2016-04-11:
I use the two functions below to switch between injection and monitor mode. They're rather long because there's some code in there for tolerating command failure (especially the set channel one) and lots of print statements.

Could you share how you got them to work simultaneously??
```
injection_mode () {
    echo "Switching $wlan_interface to inject........................................"
    ip link set $wlan_interface down
    echo "Deleting mon0...................................................."
    iw dev mon0 del 2>/dev/null 1>/dev/null
    echo "Bringing up firmware............................................."
    modprobe -r iwlwifi mac80211 cfg80211
    modprobe iwlwifi debug=0x40000
    echo "Running ip link show on $wlan_interface, looping until success............."  
    ip link show $wlan_interface 2>/dev/null 1>/dev/null
    while [ $? -ne 0 ]; do
        ip link show $wlan_interface 2>/dev/null 1>/dev/null
    done
    echo "Setting $wlan_interface into monitor mode.................................."
    iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
    mode_change=$?
    while [ $mode_change -ne 0 ]; do
        ip link set $wlan_interface down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
        mode_change=$?
    done
    echo "Bringing up $wlan_interface ..............................................."
    ip link set $wlan_interface up
    echo "Adding monitor to $wlan_interface ........................................."
    iw dev $wlan_interface interface add mon0 type monitor
    echo "Bringing up mon0................................................."
    ip link set mon0 up
    echo "Killing default wireless interface, wlan0........................"
    ip link set wlan0 down
    echo "Setting channel on mon0 to $channel_number $channel_type ............................."
    iw dev mon0 set channel $channel_number $channel_type
    channel_set=$?
    while [ $channel_set -ne 0 ]; do
        ip link set $wlan_interface down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
        ip link set $wlan_interface up
        iw dev mon0 set channel $channel_number $channel_type
        channel_set=$?
        if [ $channel_set -eq 0 ]; then
            echo "Fixed problem with set channel command..........................."
        fi
    done
    echo "Setting monitor_tx_rate.........................................."
    echo 0x4101 | sudo tee `sudo find /sys -name monitor_tx_rate`
    echo "Injection mode active!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!"
}

monitor_mode () {
    echo "Switching $wlan_interface to monitor......................................."
    echo "Bringing up firmware............................................."
    modprobe -r iwlwifi mac80211 cfg80211
    modprobe iwlwifi connector_log=0x5
    echo "Bringing down $wlan_interface ............................................."
    ip link set $wlan_interface down 2>/dev/null 1>/dev/null
    echo "Setting $wlan_interface into monitor mode.................................."
    iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
    mode_change=$?
    while [ $mode_change -ne 0 ]; do
        ip link set $wlan_interface down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
        mode_change=$?
    done
    echo "Bringing up $wlan_interface ..............................................."
    ip link set $wlan_interface up
    wlan_interface_up=$(ip link show up | grep $wlan_interface | wc -l)
    while [ $wlan_interface_up -ne 1 ]
    do
        ip link set $wlan_interface up
        wlan_interface_up=$(ip link show up | grep $wlan_interface | wc -l)
    done
    echo "Bringing down default wireless interface wlan0..................."
    ip link set wlan0 down
    echo "Setting channel to monitor on $wlan_interface to $channel_number $channel_type ................." 
    iw dev $wlan_interface set channel $channel_number $channel_type
    channel_set=$?
    while [ $channel_set -ne 0 ]; do
        ip link set $wlan_interface down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
        ip link set $wlan_interface up 2>/dev/null 1>/dev/null
        ip link set wlan0 down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set channel $channel_number $channel_type 2>/dev/null 1>/dev/null
        channel_set=$?
        if [ $channel_set -eq 0 ]; then
            echo "Fixed problem with set channel command..........................."
        fi
    done
    echo "Monitor mode active!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!"
}
```

simondchen, 2016-04-12:
transmitter:
    rmmod iwlwifi mac80211 cfg80211
    modprobe iwlwifi debug=0x40000 connector_log=0x1
    iw wlan0 set channel 64 HT20
    ifconfig wlan0 up
    echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`

receiver:
    rmmod iwlwifi mac80211 cfg80211
    modprobe iwlwifi debug=0x40000 connector_log=0x1
    iw wlan0 set channel 64 HT20
    ifconfig wlan0 up
    echo 0x4101 | sudo tee `find /sys -name monitor_tx_rate`

as you can see, the conf of receiver and transmitter is same. I use wlan0 to transmit and receive at the same time, and it works well. I don't why it does not work if I add a mon0 interface.

egaebel, 2016-04-12:
Are you able to obtain CSI information while doing this?

simondchen, 2016-04-12:
Yes!

egaebel, 2016-04-12:
Very interesting. I'll have to give this a try, although I'm afraid it won't be for a while.

Thanks for sharing!!

simondchen, 2016-04-13:
@ Have you tried, is it okay? if I change the mac address 00:16:ea:12:34:56 in random_packets.c, no CSI data can be obtained,  is the mac address hard coded in the driver, I will check it ^_^

swashah, 2016-11-11:
@simondchen : I tried your step but it gives error opening lorcon interface ....any idea?