dhalperi/linux-80211n-csitool-supplementary #163: monitor mode:send packet and receive packet at the same time

simondchen, 2016-04-11:
When I used monitor mode to transmit packets, If the wlan0 interface is down, the packets can be sent out by runnig:./random_packet 1 100 1, but if the wlan0 interface is up, no packets can be sent out. Can anybody tell me why

asifhanif1122, 2016-04-11:
Which scripts are you using to send packets with wlan0 in UP state ?

simondchen, 2016-04-11:
@asifhanif1122 I'm using ./random_packet 1 100 1 as well with wlan0 in UP state. I tried just now, and if I   run iw wlan0 set channel 64 HT20 and then ifconfig wlan0 up, packets can be sent out. Howerver, no CSI packets can be captured if with mon0 and wlan0 in UP state at the same time

simondchen, 2016-04-11:
@asifhanif1122 I succeed in sending and receiving simultaneously by using wlan0 interface to send and receive at the same time without adding a mon0 interface. Thanks a lot!

asifhanif1122, 2016-04-11:
Try to run these commands. It will probably help you. 
[image]

asifhanif1122, 2016-04-11:
You can change your channel as well. Currently in these commands 64th channel is being selected. To scan channels supported by your wifi card run this command: **_iwlist channel_** It will give you entire list of channels and corresponding frequencies.

simondchen, 2016-04-12:
Thanks for your patient explanation, a big favor!

hamaoui, 2016-06-23:
@simondchen 

could you please tell how you did it? (sending and receiving simultaneously by using wlan0 interface)