dhalperi/linux-80211n-csitool-supplementary #164: Can't compile  log_to_file.c code

GertjanEnHarald, 2016-04-14:
Hello guys,

When I try compile that .c code with "cc log_to_file.c -o log_to_file", I get the error:

'CN_IDX_IWLAGN' undeclared in proc_addr.nl_groups = CN_IDX_IWLAGN

anyone know what the issue could be?

Thanks

egaebel, 2016-04-21:
You need to utilize the make file in the netlink directory. log_to_file.c has dependencies. You can compile by running 'make' in the netlink directory, or by coming up with an equivalent compilation string by examining the Makefile.

azizshahics, 2016-04-27:
how can I apply sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat.

When I enter the above command it shows error. Command sudo log_to_file not found.
i have installed sudo in my linux kernel.

please help

egaebel, 2016-04-27:
Did you compile log_to_file? If the problem is actually with sudo then you should look on the super user stack exchange, Google, and linux forums. Not here.

haimapi, 2016-04-28:
@azizshahics 
1. cd to linux-80211n-csitool-supplementary/netlink/
2. use: ./log_to_file csi.dat

zainalabidenakhtar, 2016-05-21:
i am trying to process my .dat file in MATLAB but i get an error:
i write this command 
>> csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');
and get this error:
??? Undefined function or method 'read_bfee' for input arguments of type 'uint8'.

Error in ==> read_bf_file at 66
        ret{count} = read_bfee(bytes);

can anybody help me in this regard