dhalperi/linux-80211n-csitool-supplementary #165: Problem setting up injection mode. Device or resource busy

Mo2000, 2016-04-20:
I followed the steps shown on the injection script page: https://github.com/dhalperi/linux-80211n-csitool-supplementary/tree/master/injection

I started off by making sure my wlan device is up by using **sudo ifconfig wlan0 up** then did the following:
- I installed libcap-dev using **sudo apt-get install libcap-dev**
- I downloaded LORCONv1 by using **git clone https://github.com/dhalperi/lorcon-old.git** and then installed it using **sudo make install**
- I then compiled the injection and monitor mode files using **make**
- I then ran **./setup_inject.sh 64 HT20** on the trasnmitter, but received **command failed: Device or resource busy (-16)**

Other things to note. I disabled NetworkManager from controlling my wlan device, so I am using iwconfig to connect to access points.

What am I missing here?

Thanks

egaebel, 2016-04-21:
I found that this seems to be an issue with set channel. It will often fail for reasons that are not evident to me. I'm guessing they relate to the timing of certain things in the wireless system being turned on and off.

Anyway, a workaround I came up with is that if set channel fails to just try the steps leading up to set channel over and over again until set channel succeeds. I've been working this way constantly for the past couple months and it works every time.

Here's the code:

```
    iw dev mon0 set channel $channel_number $channel_type
    channel_set=$?
    while [ $channel_set -ne 0 ]; do
        ip link set $wlan_interface down 2>/dev/null 1>/dev/null
        iw dev $wlan_interface set type monitor 2>/dev/null 1>/dev/null
        ip link set $wlan_interface up
        iw dev mon0 set channel $channel_number $channel_type
        channel_set=$?
        if [ $channel_set -eq 0 ]; then
            echo "Fixed problem with set channel command..........................."
        fi
```

$channel_number $channel_type and $wlan_interface are just parameters to the script, $channel_number is equivalent to $1 in setup_inject.sh, $channel_type is equivalent to $2 in setup_inject.sh and $wlan_interface is just the name of the wireless interface you want to do this on (I.E. wlan0).

The expression:

`channel_set=$?`

grabs the exit status of the previous command (which was set channel) and saves it. If the exit status is 0 then the command succeeded, otherwise, it failed.

Hope this works for you.

Mo2000, 2016-04-26:
@egaebel Thank you for that post. I was able to successfully get the injection mode working now, however I now face a different problem.
I have two computers, A and B. I was able to successfully inject packets on computer B and capture them by placing computer A in monitor mode. However, I was not able to get it to work the other way around (A injects and B monitors). I was able to successfully execute the **setup_inject** and **setup_monitor_csi** scripts, however once I would start injecting on computer A, computer B simply does not pick up anything, as if no data had been transmitted. I am not sure as to why this is happening. It worked perfectly fine one way, but won't work when the computers switched roles.

Do you have any suggestions on how I would go about debugging this? I was thinking of trying another computer and placing it into monitor mode and seeing if it can pick up the injected packets from computer A.

egaebel, 2016-04-27:
@Mo2000 

I had a great deal of trouble with the 

iw dev <device> set channel <channel_number> <channel_type> 

command failing for no apparent reason. Start there. Try running the content of the scripts yourself. Command-by-command until you find the problem.

Mo2000, 2016-04-27:
@egaebel I have just spent the past few hours looking at this and I can't seem to figure out what the problem is. I manually ran the commands and made sure that the interfaces were in monitor mode and are on the same channel. I was able to run it again by having computer B inject and computer A monitor, but I could not reverse the roles. I need CSI collection from both sides for my project.

I wanted to mention that we have only one antenna connected to each of the computers' Intel 5300 NIC. Do you think this can be causing an issue?

egaebel, 2016-04-27:
@Mo2000 

I don't think antenna count would affect this. Are you using only one wireless interface on each computer?

Mo2000, 2016-04-27:
@egaebel Yes, I am.
So I just tried again for about the third time, and I was only able to capture the data on one side. I even rebooted the computers before changing them from transmitter to receiver and it still failed. I was, however, able to ALMOST get computer B to successfully capture the packets injected by computer A, however it only captured three packets then stopped. I'm not sure why. I injected 100,000 packets of size 100. Having computer A in monitor mode and B in injection mode seems to capture almost all of those 100,000 packets.

@dhalperi do you have any idea on what might be causing this?

I appreciate everyone's help here. Thank you

Hirokazu-Narui, 2016-11-02:
Today, I fixed this kind of error is disappeared to change some shell script.
In "setup_inject.sh"  -> "ifconfig mon0 up" and "iw mon0 set channel $1 $2" are up side down.
In "setup_monitor_csi.sh" -> "ifconfig mon0 up" and "iw mon0 set channel $1 $2" are up side down.

After that, my setup running.

(Reference)
https://www.spinics.net/lists/linux-wireless/msg127023.html

lidongning, 2017-06-22:
./setup_inject.sh 64 HT20 :  command failed: Device or resource busy (-16)  how to edit ???  thank you @Mo2000

lidongning, 2017-06-22:
./setup_inject.sh 64 HT20 : command failed: Device or resource busy (-16) how to edit ??? thank you @Mo2000   I don't understand what you said? @Hirokazu-Narui

Hirokazu-Narui, 2017-11-27:
@lidongning 
Original code ( linux-80211n-csitool-supplementary/injection/setup_inject.sh):
line10 : ifconfig mon0 up
line11 : iw mon0 set channel $1 $2

My code:
line10 : iw mon0 set channel $1 $2
line11 : ifconfig mon0 up

I fixed "linux-80211n-csitool-supplementary/injection/setup_monitor_csi.sh" in the same way.

(Reference)
https://www.spinics.net/lists/linux-wireless/msg127023.html