dhalperi/linux-80211n-csitool-supplementary #166: Where is the CSI matrix extracted from the 5300NIC, before FFT of after?

zjulsp, 2016-04-25:
The receiver is as follow. Then which part is the csi matrix extracted from? The candidates are "position1", "pos2","pos3" and "pos4", or maybe otherwhere. 
[Antenna]->[ADC and Down-RF]->[AGC Signal detect]->**position1**->[Frequency synchronization]->**pos2**->[Fine time synchronization]->**pos3**->[FFT]->**pos4**->[...]
Thank you.

dhalperi, 2016-04-25:
I don't think your diagram adequately captures all possibilities of receiver chains, especially with the complexity in real hardware. The CSI is definitely extracted without the finest possible time synchronization, however, as you can see by looking at measurements from one packet to the next.

zjulsp, 2016-04-28:
Hi Daniel. Thanks for the quick reply.☺  @dhalperi
Consider the process of equalizer.
- Let **Y**[k](k=1,2...48) be the the FFT of the received LTF symbol after frequency correction, well,  just consider the data-subcarriers.  
Let **H**[k] be the frequency response of the channel, and **X**[k] be the transmitted LTF before IFFT which has already been assigned by IEEE 802.11. 
We know that **Y**[k]=**H**[k]***X**[k]+**Z**[k], where **Z**[k] is the noise. Then we can estimate the channel as follow,
**H**_estimated_=**Y**[k]/**X**[k];
- Is the csi-matrix we get from 'csi_entry.csi'  the **frequency** response of the channel, i.e. H[k] I mentioned before? If yes, is the csi-matrix Intel5300NIC provides able to reflect the multi-path feature of the channel?
Again, Thanks a lot.