dhalperi/linux-80211n-csitool-supplementary #168: How to find out channel bandwidth?

awelkie, 2016-05-01:
How can I determine the channel bandwidth for recorded data? For example, the data in 'matlab/sample_data' has three different values for the `rate` field: [0x100, 0x108, 0x110]. I'm assuming that this data is from a 20MHz channel because the [RATE_MCS_HT40_MSK](https://github.com/dhalperi/linux-80211n-csitool/blob/csitool-3.13/drivers/net/wireless/iwlwifi/dvm/commands.h#L309) bit flag is set to 0 in these values. Is this correct?

asifhanif1122, 2016-05-04:
See this snapshot to get idea about settings while transmitting data in monitor mode.
[image]

dhalperi, 2016-05-07:
@awelkie yes -- this is right. You can also read the FAQ to confirm this.

closed 2016-05-07 by dhalperi