dhalperi/linux-80211n-csitool-supplementary #169: How the Phase and Amplitude in CSI are calculated per packet

soltanaghaei, 2016-05-06:
Hi, I would like to know how the phase and amplitude are calculated for each packet. The phase shift calculation requires a reference and I supposed that the beginning of packet signal is considered as the reference time. However, there is no synchronization scheme to recognize that the received signal relates to the new packet. Then,  just a small time offset changes the calculated phase shift. I would like to know how accurately the phase and amplitudes are calculated in CSI by understanding the algorithm of this calculation. I would be grateful if you can share a document with me to explain this.

In addition, we finally have one CSI value per frequency, while multiple bits of the packet (or preamble of the packet in which CSI is calculated) are sent in each subcarrier. Are CSI values calculated per bit and then averaged in each frequency?

Thank you in advance for your help :)

soltanaghaei, 2016-05-12:
Are my questions clear enough? What is the problem that I didn't have any reply or comment after 6 days? The summary of my questions are:

1) How are amplitude and phase values of each subcarrier calculated while multiple symbols of the preamble are transferred in each subcarrier? are they averaged or used as new equations for CSI calculation?

2) How much the phase offset caused by packet detection delay affects the accuracy of phase values? Can I assume that the phase error is the same for all the packets or for all the subcarriers? 

I really need the answer of these two question, read all the documents, thesis and papers in CSI, but couldn't figure out these two points :(

yangguang010, 2018-11-21:
Hello! I want to ask you how you get the corresponding frequency offset through the collected CSI information (plural). Thanks!