dhalperi/linux-80211n-csitool-supplementary #170: Is it required to connect all three antennas in the receiver to get the correct CSI values>

soltanaghaei, 2016-05-06:
Hi, my experimental laptop only has two antennas, but I changed the network card to a 5300 with three antenna and so no antenna is connected to the third slot. I would like to know If I run the CSI tool on this network card, does it cause any error in the calculations? 

Actually, this question raised for me based on the requirements mentioned in the FAQ (number 5), as it says the receiver physical configuration such as the correct number of antennas should be satisfied.

asifhanif1122, 2016-05-07:
It will not cause any trouble. You will find junk values for 3rd antenna CSI data in .dat file .

soltanaghaei, 2016-05-12:
Thank you very much for your reply. What if the laptop as the transceiver in Injection mode has only two antennas?

luxiangx, 2016-12-30:
You can set moniter_tx_rate to match the two antennas, it would be ok. I think.