dhalperi/linux-80211n-csitool-supplementary #171: How can i convert CSI complex to SNR without matlab

BrunoSoares-LABORA, 2016-05-06:
Hi all,

I have read the matlab code and topic 6 of FAQ but i can't understand how can i convert the CSI complex (real and imag) to SNR value. Can you help me?

asifhanif1122, 2016-05-10:
Bruno use this matlab function to get SNR; 
db(get_eff_SNRs(csi), 'pow')
Make sure you have obtained .m file of this function from forum and is in correct directory.

BrunoSoares-LABORA, 2016-05-10:
@asifhanif1122 

Thanks for supporting, but i want to convert the complex number not using matlab. i also tried to understand what this function makes with csi complex number but without success, maybe because i have no experience with matlab. Can you explain it for me?

asifhanif1122, 2016-05-11:
See this simple snippet of code;
[image]

When you successfully get CSI file , then its dimension of one packet ( i.e csi{1} ) will be 1x3x30 . **Squeeze** function converts this dimension from 1x3x30 to 3x30. Now  **abs** function calculates magnitude of this complex matrix. **db** function converts scale to deciBel scale so that very small values may become easy to view. If you still find difficulty, feel free to discuss.

azizshahics, 2016-05-11:
@BrunoSoares-LABORA  I can help you. email me at azizshahics@yahoo.com

asifhanif1122, 2016-05-11:
@azizshahics It would be in beneficial if we discuss issues on this forum rather than seeking help through emails. If you explain solution openly, everyone will be able to benefit from your precious comments. Don't mind my opinion; it is in everyone's interest.

BrunoSoares-LABORA, 2016-05-11:
@asifhanif1122 

What i don't understand is:
- **abs** and **db** functions iterate over all 90 (3x30) complex number?
- How **abs** function calculate the magnitude of a complex number?

Thanks for supporting.

asifhanif1122, 2016-05-11:
Okay. Functions in Matlab are designed to work on matrices. When you pass name of a matrix to any of the above mentioned functions ; they will perform required operation on each element present in matrix. For example if you want to find magnitude of each complex element in matrix, **abs** function will return a matrix with same dimension but data inside will be magnitude of input complex matrix. 

complex number = a + i*b ;
magnitude of complex number = square_root (  a^2 + b^2 ) ;

**abs**   will perform this operation on each element of input matrix and output matrix will contain same number of elements(magnitude of complex numbers) as it was in input matrix. 
I will suggest you to  watch some basic tutorials on Youtube for Matlab programming. I hope you got my point. :)

azizshahics, 2016-06-15:
@zainalabidenakhtar  take IFFT of the SNR vs Subcarriers.
you will get data in time domain.
If you face problem.let me know

lixiaohai, 2016-07-08:
@azizshahics 
I do as you said in matlab.Here is my result.
This is the SNR vs Subcarriers.
[image]
And this is the result after i take IFFT of the SNR vs Subcarriers.
[image]
Is there anything wrong?
It seems that a paper I have readed has a different result.
This is the link.
http://www.cse.ust.hk/~kwinson/papers/3_CSI-based%20Indoor%20Localization_TPDS.pdf
And this is his data in time domain.
[image]
THANK YOU! Look for your favorite reply.

tahmidzbr, 2016-11-11:
@brunosoaresds @asifhanif1122 @azizshahics @zainalabidenakhtar @lixiaohai 
Can anyone please explain why taking the IFFT of the SNRvsSubcarrierIndex gives the SNR in time domain?

azizshahics, 2016-11-12:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } Take all the amplitudes of specific subcarrier ( say subcarrier #12) and plot it.It would give you CFR time history.you can scale the time axis accordingly.

linkzhao, 2016-11-14:
@azizshahics how to take IFFT of the SNR vs Subcarriers,do you mind share your matlab code ?thanks!

brunosoaresds, 2016-11-15:
@linkzhao @tahmidzbr 
I Think the data collected from this tools is already in time domain, i.e, you don't need to put data into IFFT function to get the SNR of some subcarrier in time. All you need to do is get the data of some subcarrier for all packets in some time and plot then. Remember to put the complex numbers into db(abs(get_csi_scaled(**data**))) to get the SNR.

If your question was not answered correctly, i will need more explanation :)

tahmidzbr, 2016-11-15:
@brunosoaresds thanks for your answer i have some questions from your answer
1) db(abs(get_csi_scaled(csi)) is to get SNR for any 1 packet right? How is this even done? Like each CSI matrix has entries in cartesian form, and if we take absolute value of it isnt it the amplitude? So why is this the SNR and not amplitude?
2) How is the CSI matrix arranged? Are the columns receving antennas A C B? or it depends on RSSI of each receiving antenna?
3) I didnt actually understand how to do the time domain?
Thanks in advance

brunosoaresds, 2016-11-15:
@tahmidzbr 

1. Right, but remember that **packet** is a csi struct with some meta-data and csi matrix including RXxTXx30 complex numbers. You will need to read about this functions but basiclly get_csi_scalled() function will remove AGC gain and thermal noise from the magnite of each trace of CSI packet. abs() function will convert the complex number (the matrix) to magnitude (amplitude) and db() function will convert the magnitude to SNR in dbm.
**"So why is this the SNR and not amplitude"** - Remember that the CSI only have the signal amplitude including the noise, so.. the question is:  "How much of this amplitude is not a noise?" because the trace include both and this is why this data is a SNR.
2. I think i don't understand this question. but the matrix is Transmitter x Reveiver x Subcarriers and yes 1 = A, 2 = B and 3 = C.
3. Well, since we have each packet inside time domain, we can extract the amplitude of each of then and plot this amplitude wich is the amplitude in time domain. All you need to do is get the amplitude of each packet distributed in some time period and plot then in sequence.

tahmidzbr, 2016-11-15:
@brunosoaresds Many thanks for your answer. Do you know where i can read about these in more detail? Can you please suggest a good reference?

brunosoaresds, 2016-11-15:
@tahmidzbr "Fundamentals of Wireless Communications from David Tse and Pramod Viswanath" and the IEEE 802.11n pattern helped me a lot.

tahmidzbr, 2016-11-15:
@brunosoaresds thank you buddy. Im doing my research work at the Univ of British Columbia, and if you want we can discuss further into these area. My email is tahmidzbr@ece.ubc.ca
Have a good day.

linkzhao, 2016-11-16:
it is not one problem, i want to get something with one csi packet

swashah, 2016-11-19:
@brunosoaresds : I am collecting CSI on 02 laptops in minitor mode and third laptop is injecting packets. I have tried to keep the distance between 02 antenna to be less then half of wavelength. But when I collect csi and takes its abs(csi) I see that subcarrier 9:19 have similar amplitude on both machines and logically they should be similar. But other subcarriers are not similar. Do you have any idea why is this happening.Thanks

brunosoaresds, 2016-11-20:
@swashah sorry, i don't know how to explain that. Do you have tried other distances between the antennas?

swashah, 2016-11-20:
@brunosoaresds : No I have not tried other distances. Another problem I am facing is that when I record CSI on two machines , 02 measurements are not aligned (or synchronized). Do you have any idea how to synchronize data on 02 machines using time stamp or any thing else?

brunosoaresds, 2016-11-21:
@swashah I think it's not possible to synchronize it. If i undestand correctly you want that the 2 nootebooks receive the packet at the same nanosecond right? I think you can't do it because naturally the packet will be received in diferent times by the antennas, another issue is the clock synchronization between the notebooks, so i think that is not possible or its not EASY. Can i ask you why you need it?

tahmidzbr, 2016-11-21:
@brunosoaresds  Hello again, when you said we have each packet in time domain, i dont quite understand how to find the time? Like say if i ping 100 packets/s does it mean every 0.01s i have one packet? so i can extract the amplitude of this packet and plot it every 0.01s?
Is this what you meant?

Also another question was Im using AP as my Tx and laptop with 5300 as Tx. Sometimes i see that my no. of transmit antennas and no. of receive antennas just change. Like i dont change environment or move them, but i see Nt and Nr varying. Sometimes my CSI is 2x3x30 sometimes its 1x2x30, i.e. Nt and Nr keeps varying. Do you know why?

swashah, 2016-11-21:
@brunosoaresds : I have seen few works where they have done this type of synchronization but I am not getting hold how to do it. I want to show that when two antennas are close enough they will have very similar CSI. I am obtaining similar CSI but one copy is slightly delayed so I want to align these 02 copies.

kahnreaz, 2016-12-16:
Hi, indeed its a great discussion. I learnt a lot and got cleared up some of my own confusions. Now @swashah I am also working on the similar. So, I also need synchronization to check channel reciprocity. It would be a real help if soemone could shed some lights on how to synchronize it. perhaps I can bear with nano second delay. @swashah  could you help me about plotting csi frm botm node. actually in our setup we dump csi as txt file. since there are some controllers involved on top of the original csi tool driver.

sfehlandt, 2017-03-20:
Hi @asifhanif1122, I am confused. Where can I find this db() function for Octave?
Is there any reason this function is not already included in the repo?

I ended up implementing it myself based on the documentation of the Matlab db() function.
I have commited it to a fork copy of the repository, so maybe if it makes sense I can do a pull request to add my implementation of this function to the matlab folder.

tfsanam, 2017-04-28:
@brunosoaresds According to your suggestion:

> I think the data collected from this tool is already in time domain, i.e, you don't need to put data into IFFT function to get the SNR of some subcarrier in time. All you need to do is get the data of some subcarrier for all packets in some time and plot then. Remember to put the complex numbers into db(abs(get_csi_scaled(data))) to get the SNR.

I have arranged a matrix where each row indicates a packet from the csi struct and in columns I have the CSI amplitudes for each tx-rx pair for each subcarrier.

So my plot looks like this:
[image]
 here x axis indicates the number of packets, totatl 693 packets.
I want the x axis as time. Can you help me how can I get x axis as time?