dhalperi/linux-80211n-csitool-supplementary #174: i am using Lenovo V570c laptop. when i try to install intel 5300 card in slot a message appears that hardware not supported. is there any option I can use this card using some change in hardware or software?

zainalabidenakhtar, 2016-05-15:


zainalabidenakhtar, 2016-05-15:
[image]

zhangshaohu, 2016-05-16:
I tried a couple of laptops. I think you have to use the laptop with Intel WiFI 5300 card

asifhanif1122, 2016-05-16:
If you have Dell laptop, it will work.

zainalabidenakhtar, 2016-05-17:
finally, I changed my laptop

asifhanif1122, 2016-05-17:
Please share your experience. Which laptop worked for you?

vinatby, 2016-05-18:
http://www.intel.com/Assets/PDF/general/5300_laptop_list.pdf

Check out this link..

soltanaghaei, 2016-05-19:
@vinatby @asifhanif1122 I had exactly the same issue and still looking for a laptop to support 5300 card. All the laptops listed in this link are old and not available any more. Does anybody can introduce me an available laptop with 3 antennas that support 5300, so that I can buy?

libing64, 2016-05-20:
@soltanaghaei I am using Intel NUC Computer and second-hand Thinkpad T400, both work with Intel 5300 wireless card. You can have a try.

vinatby, 2016-05-20:
Asus s550 also works fine in my case. but only two antennas..
I guess some of the advance laptops(not listed here) would be good enough for this card.

soltanaghaei, 2016-05-20:
Thanks all for your suggestions. I finally found Dell xps M1330 which has three built-in antennas, but the wifi card is a full size mini-PCI. Although you can still use the half size wifi card in this case, the third antenna's wire is too short to reach to the card. Therefore, if you would like to use Dell xps with three-antenna connection, be aware to buy a full size 5300 card.

soltanaghaei, 2016-05-20:
@libing64 Good suggestion. Could you use three antennas with NUC? the models that I found all have two antenna ports.

libing64, 2016-05-21:
@soltanaghaei Yes, I am using three antennas with NUC. The third antenna is bought from taobao.

dpward, 2016-05-24:
[Here's a description](http://www.thinkwiki.org/wiki/Problem_with_unauthorized_MiniPCI_network_card#Problem_description) of the issue you are encountering with your Lenovo laptop.

(I'm only referring to the problem description. Using any other information on that page is entirely at your own risk and may void any manufacturer's warranty.)

closed 2016-05-24 by dpward

zainalabidenakhtar, 2016-05-31:
useful information @dpward   thanks