dhalperi/linux-80211n-csitool-supplementary #176: Phase Sanitization

asifhanif1122, 2016-05-17:
Hi everyone,
I am working on phase sensitization algorithm but facing some computational difficulties. In the following snap-shot phase is being sanitized using linear transformation. I am not able to understand values of  _**ki**_ that are actually frequencies of sub-carriers in this transformation. If anyone has idea about exact values of frequencies of sub-carriers for 20MHz bandwidth channel, kindly let me know. 
@dhalperi 
[image]

vejja, 2016-05-17:
My 2 cents
1. ki sounds like an index (1, 2, 3, ...) not a frequency. Otherwise the formula would not be homogeneous
2. the frequency fi of subcarrier i is fi = f0 + df * i, where f0 is the center frequency (2.437 GHz here), df = channel bandwidth / nb subcarriers (20Mhz / 64 = 312.5 KHz here) and i is the subcarrier index sitting in the range [-28, +28]

So for instance the frequency of subcarrier with index -28 is : 2.437GHz - 28 * 312.5KHz = 2.428250 GHz

asifhanif1122, 2016-05-17:
thank you @vejja  Can you look into this algorithm. I tried to implement it in Matlab but got unexpected results. Does variable **n** show maximum sub-carrier index? If yes , what will be its value 30 or +28 ?

vejja, 2016-05-17:
At first sight it looks like n is the number of samples you have collected, i.e. it could well be 1,000 judging from the plot on the left side.
Can you send a link to the algorithm you are using ?

asifhanif1122, 2016-05-17:
This is the link for paper in which phase is being sanitized using above mentioned algorithm.

http://tns.thss.tsinghua.edu.cn/~cswu/files/ICPADS14_PADS_paper.pdf
Page 4 , ( Equation 5 , 6 and 7 )

vejja, 2016-05-17:
Ok so n is *not* a number of samples but n is 30
They say : 
```
ki denotes the subcarrier index (ranging from -28 to 28 in IEEE 802.11n) of the ith subcarrier
```
So as I understand
- i ranges from 1 to 30
- n equals 30
- ki ranges from -28 to 28

For instance : the 1st subcarrier has a subcarrier index k1 = -28, the 2nd subcarrier has a subcarrier index k2 = -26, etc.

azizshahics, 2016-08-01:
@asifhanif1122  HI
      I would like to analyse the CSI measured data into phase response.
I would really appreciate if anyone can help.

regards

mazilaile, 2016-11-28:
@asifhanif1122 @azizshahics ，if you  got expected results? Can you help me ??

asifhanif1122, 2016-11-28:
I made Matlab function for CSI phase sanitization. You can use it. It takes an array of 30 CSI phase values and returns output array containing sanitized phase values.

[sanitize_phase.zip](https://github.com/dhalperi/linux-80211n-csitool-supplementary/files/615906/sanitize_phase.zip)

mazilaile, 2016-11-28:
@asifhanif1122 thank you for your codes. I made a mistake , ki=-28,26....  
From you code,I know ki=[-15:1,1:15], but the result is not so well. 
[image]

azizshahics, 2016-11-29:
@mazilaile  I have run the above code but could you please indicate how did you exactly plot the above figure?
Could you please share the matlab commands?

azizshahics, 2016-11-29:
[image]

azizshahics, 2016-11-29:
I need matlab command for above plot.I know to run the function provided by Asif. but unable to plot the particular figure

azizshahics, 2016-11-29:
@mazilaile Could you please help

paulkirkland, 2017-01-11:
To plot the above, it is just using polarplot in Matlab for the values of (rho) - According to Mathworks...

I am unsure about the code though all the Matrices and Vectors are transposed or not aligned with the code as is. What are you using an input phase, the IQ value of phase from the complex number csi, as a -- number of data points x 30 (subcarriers) ?

elaheriahi, 2019-09-24:
> I made Matlab function for CSI phase sanitization. You can use it. It takes an array of 30 CSI phase values and returns output array containing sanitized phase values.
> 
> [sanitize_phase.zip](https://github.com/dhalperi/linux-80211n-csitool-supplementary/files/615906/sanitize_phase.zip)

I want to use your code for 256 phase subcarriers, I modified your code accordingly. but I got an unexpected result. can you please help me? you can see the unwrapped phase, and the sanitized phase in the
[image]

lagharimanzar, 2019-09-24:
Hello

IWL5300 NIC only reports 30 out of 64 sub-carriers for 20 MHz and 30 out of 128 for 40 MHz. Why are you trying to get 256 sub-carriers whether it is right or not? But you can achieve 64 and 128 sub-carriers by interpolation.

elaheriahi, 2019-09-24:
> 
> 
> Hello
> 
> IWL5300 NIC only reports 30 out of 64 sub-carriers for 20 MHz and 30 out of 128 for 40 MHz. Why are you trying to get 256 sub-carriers whether it is right or not? But you can achieve 64 and 128 sub-carriers by interpolation.

I am using another tool for CSI extraction. that´s why the bandwidth is 80 MHZ, with the FFT length of 256. I have interpolated already the phase for some null subcarriers. and I want to use the function for this number of subcarriers. Do you have any idea how to do the sanitzied phase correction for this bandwidth directly?

lagharimanzar, 2019-09-25:
No.

Im using 5300 only. 

I have made function for 20 and 40 MHz. if your function is working with 20/40 then should work for 80 MHz just adding more sub-carrier indices in it

elaheriahi, 2019-09-25:
> 
> 
> No.
> 
> Im using 5300 only.
> 
> I have made function for 20 and 40 MHz. if your function is working with 20/40 then should work for 80 MHz just adding more sub-carrier indices in it

Can you please share your function here? Maybe I am doing a part wrong.