dhalperi/linux-80211n-csitool-supplementary #178: unable to access ip address from AP using dhclient and also need guidance for encryption key

vinatby, 2016-05-17:
Hello all,
first of all, thank you for this awesome device and reference.. I am a beginner in linux..
I have somehow managed to reach upto the point where I have to reload driver with CSI logging method ( Second last in 5th in installation guide) . But I am struggling with getting AP's  IP address using command dhclient wlanx (wlan1 in my case with previleges). Also I can not set my encryption key off as you have suggested in FAQs. Actually I do not understand how that works. is it default 'on' from my router ? ( do i have to change it from there or I can change it from my laptop? )

When I use command dmesg, it sometimes give me errors like wlan1 is deauthenticating by local choice automatically (reason=3).. I do not know how to solve this problem, however it happens sometimes only.. kindly help me with this issues and if possible provide some references for some fast paced understanding.

** my router supports 802.11n, I have loaded modified drivers properly on my 5300 card, I have been playing with iw, network-manager, modprobe and lsmod like commands from references on internet

any help would be highly appreciated. let me know if you want additional info. of my setup.

vinatby, 2016-05-17:
When I write command lspci -k  , I get Network controller: Intel comp..ultimate N WiFi link 5300.... Kernel Driver in use : iwlwifi

and after that I typed
dmesg | grep firmware    , I received output as follows

iwlwifi .....:Direct firmware load for iwlwifi-5000-5.ucode failed with error -2
                Request for firmware file 'iwlwifi-5000-5.ucode' failed.
.........................above line three times.......
........................
iwlwifi 0000:03:00:0 new firmware can be obtained from http://www.intellinuxwireless.org/.
iwlwifi ....................  loaded firmware version 8.24.2.12 op_mode iwldvm 

Could this be problem ? I think kernel is using iwlwifi driver and according to my dmesg command it tries to load iwl one but fails and end up loading iwldvm.
If yes, how could it be solved ( please little explicitely).

vinatby, 2016-05-18:
I am using router motorola surfboard SBG6580 which is compatible with 802.11n , I have set encryption to off and freq to 5 GHz. My Laptop is Asus S550 installed with Ubuntu 14.04.1 version and NIC 5300 is already installed here.. output of my uname -a is as follows ..

Linux tug14467-Q550LF 4.2.0-35-generic #40~14.04.1-Ubuntu SMP Fri Mar 18 16:37:35 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

I am sorry these posts should be chronologically reversed .       Let me know if you need additional info.

zhangshaohu, 2016-05-24:
First of all, set up your router without encryption.
If your installation is correct 
run 
wens@wens:~$ sudo modprobe -r iwlwifi mac80211
wens@wens:~$ sudo modprobe iwlwifi connector_log=0x1
wens@wens:~$ 
wens@wens:~$ iwconfig
eth0      no wireless extensions.

lo        no wireless extensions.

wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
wens@wens:~$ iw dev wlan0 link                          // no WiFi connection 
Not connected.
wens@wens:~$ sudo ifconfig wlan0 up
wens@wens:~$ sudo ifconfig eth0 up
wens@wens:~$ sudo iwconfig wlan0 essid TP-LINK_DD9C  //connect with your router
wens@wens:~$ sudo dhclient wlan0
wens@wens:~$ sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat

AFTER THIS STEM, OPEN ANOTHER TERMINAL 
TYPE:
wens@wens:~$ ping ip

vinatby, 2016-05-24:
got it...! Thank you very much.

dpward, 2016-05-24:
Encryption is not supported in the modified firmware (see FAQ 3).

closed 2016-05-24 by dpward