dhalperi/linux-80211n-csitool-supplementary #179: why CSI magnitude values are more consistent compared to the CSI amplitudes?

soltanaghaei, 2016-05-18:
Hi, does any body know why the CSI magnitude/SNR values are more consistent compared to the raw scaled CSI amplitudes over sequential packets for all the subcarriers? Basically I compared the two graphs using plot(db(abs(squeeze(csi).'))) and plot(db(real(squeeze(csi).'))) and the second graph has much more variation over time.

zainalabidenakhtar, 2016-06-15:
@soltanaghaei can you please share any code to plot with respect to time because i also want to plot CSI with time instead of subcarrier index but not getting correct result may be i am doing some mistake. thanks in advance

azizshahics, 2016-06-15:
@zainalabidenakhtar  Please consider following code for time domain analysis

csi_trace=read_bf_file('june\p1b.dat');       %%% reads the CSI file p1b.dat
    csi_entry = csi_trace{1}                            %%% trace the 1st packet of CSI data [each packet 
                                                                      %%% contains SNR vs 30 subcarrier information]
csi=get_scaled_csi(csi_entry)
    db(get_eff_SNRs(csi),'pow')
        
    ya=(db(abs(squeeze(csi).')));         %%% ya represents SNR vs 30 subcarreirs for 3 Rx Antennas
    
      antenna1a=ya(:,1);                        %% Rx antenna A
    antenna2a=ya(:,2);                              %% RX antenna B
    antenna3a=ya(:,3);                              %% RX antenna C

%% Time Domain [ X axis extracting code ]
i=1:30;
   n=30; % number of subcarriers
   fop=20*10^6; % 2.43 GHz operating frequency of wifi router
   t=[(fop/2)/n]^-1; % formula taken from mathworks
   di=t*(i-1);

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

   power_time_domain=20*log10(abs(ifft(antenna2a)));     %% Y axis power plot (same as SNR)
plot(di,power_time_domain)

zainalabidenakhtar, 2016-06-20:
@azizshahics thanks a lot for your favour

zainalabidenakhtar, 2016-06-20:
how the value is 'i' is selected? is this same as subcarriers?

zainalabidenakhtar, 2016-06-20:
when I run this code I actually get this error:

Subscript indices must either be real positive integers or logicals.

Error in time_domain_plot (line 32)
di=t(i-1);

azizshahics, 2016-06-20:
yes its the same as sub-carrier number.

ogutiann, 2016-10-20:
@zainalabidenakhtar where you able to solve this error?