dhalperi/linux-80211n-csitool-supplementary #184: how to process CSI dat file in MATLAB

zainalabidenakhtar, 2016-05-23:
I am trying to process CSI data in MATLAB. I took my data file but got the same problem so I got sample file that there may be problem in my data file but same error getting

>> csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');
 ??? Undefined function or method 'read_bfee' for input arguments of type 'uint8'.

Error in ==> read_bf_file at 66
 ret{count} = read_bfee(bytes);

>> mex read_bfee.c

Select a compiler: 

[0] None 

Compiler: read_bfee
 Please select from 0-0 
 Compiler:
when I select 0 it says no compiler selected. any suggestion how to get resolved this issue?
I am using MATLAB 64-bit with windows 10 of 64-bits. I also tried the same on windows 7 with 32-bit MATLAB. am I using wrong command? please suggest

vinatby, 2016-05-23:
are you sure you are in the location where this functions are saved ..
if not you probably have to add that path or go there in matlab..( undefined function )

zainalabidenakhtar, 2016-05-25:
@vinably thank you, i was using windows 64-bit but this was not supporting some file. I tried it using 32-bit windows and problem is resolved.

closed 2016-06-07 by zainalabidenakhtar