dhalperi/linux-80211n-csitool-supplementary #186: No CSI logging after ping

ekuno90, 2016-05-25:
Hi, all.
Thank you for sharing your awesome tool.
I searched a lot of debugging and followed them but.. 
when I do the last step---"log_to_file csi.dat"and"ping AP IP " at another terminal, I can't capture any data.

Here are some codes, I tried.
-----------------------------------------------------------------
moon@moon-P34V4:~$ sudo modprobe -r iwlwifi mac80211
[sudo] password for moon: 
rmmod: ERROR: missing module name.
modprobe: FATAL: Error running remove command for iwlwifi
moon@moon-P34V4:~$ sudo modprobe iwlwifi connector_log=0x1
moon@moon-P34V4:~$ iwconfig
eth0      no wireless extensions.

wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=0 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          
lo        no wireless extensions.

moon@moon-P34V4:~$ iw dev wlan0 link
Not connected.
moon@moon-P34V4:~$ sudo ifconfig wlan0 up
moon@moon-P34V4:~$ sudo ifconfig eth0 up
moon@moon-P34V4:~$ sudo iwconfig wlan0 essid TEST
moon@moon-P34V4:~$ sudo dhclient wlan0
moon@moon-P34V4:~$ sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat
^CCaught signal 2
--------------------------------------------------------------------------------------------------------------
And ping command at another terminal.
--------------------------------------------------------------------------------------------------------------
moon@moon-P34V4:~$ ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=2.95 ms
...
64 bytes from 192.168.0.1: icmp_seq=402 ttl=64 time=2.14 ms
64 bytes from 192.168.0.1: icmp_seq=403 ttl=64 time=2.94 ms
64 bytes from 192.168.0.1: icmp_seq=404 ttl=64 time=2.94 ms
^C
--- 192.168.0.1 ping statistics ---
404 packets transmitted, 404 received, 0% packet loss, time 403375ms
rtt min/avg/max/mdev = 1.218/3.437/25.145/1.803 ms
-----------------------------------------------------------------------------------------------------------

I can't capture CSI data and csi.dat is empty.
So I tried to figure out where I did wrong. here are some codes.
-----------------------------------------------------------------------------------------------------------
moon@moon-P34V4:~$ uname -a
Linux moon-P34V4 4.2.0-36-generic #42~14.04.1-Ubuntu SMP Fri May 13 17:27:22 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
moon@moon-P34V4:~$ iw dev wlan0 link
Connected to 00:26:66:6d:d0:6c (on wlan0)
	SSID: TEST
	freq: 2467
	RX: 2473054 bytes (14741 packets)
	TX: 191996 bytes (1525 packets)
	signal: -15 dBm
	tx bitrate: 65.0 MBit/s MCS 7

	bss flags:	short-slot-time
	dtim period:	1
	beacon int:	100
moon@moon-P34V4:~$ sudo iw dev wlan0 scan
BSS 00:26:66:6d:d0:6c (on wlan0) -- associated
	TSF: 1367150066 usec (0d, 00:22:47)
	freq: 2467
	beacon interval: 100
	capability: ESS ShortSlotTime APSD (0x0c01)
	signal: -18.00 dBm
[... 133 lines cut ...]
[  644.508987] wlan0: authenticated
---------------------------------------------------------------------------------------------------------------
And now Im checking these questions.

1.    Do you have the pings being received by both endpoints successfully?
yes, I think 
--- 192.168.0.1 ping statistics ---
404 packets transmitted, 404 received, 0% packet loss, time 403375ms

2.   Are the packets being transmitted using 802.11n HT rates?
I checked my router to be changed n-only mode. but How can I sure my router transmite HT rates?

3.    Did you install the correct firmware?
yes, I think from
[  618.480392] iwlwifi 0000:04:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm

4.    Did you load the driver with the correct flags?
moon@moon-P34V4:~$ sudo modprobe -r iwlwifi mac80211
moon@moon-P34V4:~$ sudo modprobe iwlwifi connector_log=0x1

5.    Did you re-compile and run log_to_file?
moon@moon-P34V4:~$ sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat

Any help would be highly appreciated. Let me know if you want additional info. of my setup.
Thank you.

Carll7, 2016-07-12:
Did you manage to solve this problem ?

tahmidzbr, 2016-10-18:
@ekuno90 hi did you solve the problem? Im having the same problem. Can someone please help?

paulkirkland, 2016-12-14:
Did you ever solve this issue, i have a similar issue with the network AP i am trying to set up but on a generic WiFi Guest AP it work great every time???

mojtabaaghayi, 2019-06-03:
How i read csi data in matlab ?
Sm.aghayi@gmail .com

mojtabaaghayi, 2019-06-03:
How i read csi data in matlab ?
sm.aghayi@gmail .com