dhalperi/linux-80211n-csitool-supplementary #189: How to Plot CSI Data in to Time domain

azizshahics, 2016-05-29:
Hello Everybody,
                               Can anyone indicate how to plot CSI data into time domain?

thank in advance.

zainalabidenakhtar, 2016-06-15:
@azizshahics i also want to plot csi data in time domain but could not do so if you find any command/code to plot in time domain then please share

azizshahics, 2016-06-15:
its easy.take IFFT (inverse fourier transform) of the data you get in frequency domain (i.e the SNR vs 30 subcarrier plot)

zainalabidenakhtar, 2016-06-16:
thanks @azizshahics I try

ersammah, 2016-07-25:
@azizshahics can you mention the MATLAB code that can convert the CSI (CFR) into CIR!!

sunyu1122, 2016-12-12:
@azizshahics i take IFFT,but i cannot obtain the initial time in the time domain

azizshahics, 2016-12-12:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; }  Please plot ith subcarrier vs packet number Ith subcarrier yaxis , packet number x axis
Say you select subcarrier number 10.plot amplitudes of 10th subcarrier vs total number if packets received.

sunyu1122, 2016-12-13:
@azizshahics [image]
I cannot obtain the initial time in the time domain , i need the help about it ,thanks.

azizshahics, 2016-12-14:
For time domain, select a subcarrier, say subcarrier # 10, plot packet number on x-axis and corresponding amplitude on y-axis will give you time plot.

azizshahics, 2016-12-14:
@sunyu1122 For time domain, select a subcarrier, say subcarrier # 10, plot packet number on x-axis and corresponding amplitude on y-axis will give you time plot.

azizshahics, 2016-12-14:
@sunyu1122  no need to take fft

xiaolingu, 2016-12-26:
@azizshahics Sorry,how to plot the above graph？every article says to apply the ifft for converting to CIR.Can you describe the detailed steps？thanks for ahead.
.

azizshahics, 2016-12-26:
@xiaolingu  Yes you have to take IFFT to get CIR, but in this particular case you need to plot all the amplitudes of a particular subcarrier (i.e. subcarrier # 12) against packet index.This is right way to plot the time domain graph.

Should you have any questions, please let me know.

xiaolingu, 2016-12-26:
@azizshahics Would you explain a simple instance how a CSI vector of 30 subcarriers is converted to a CIR?Can you give me a email ? This public space couldn't be suitable in talk too often.

azizshahics, 2016-12-26:
@xiaolingu azizshahics@yahoo.com, send me an email.

wawahahahahahaha, 2017-03-03:
@azizshahics @xiaolingu excuse me?how to plot the CIR and CFR graph？every article says to apply the ifft for converting to CIR.Can you describe the detailed steps？thanks for ahead.

wawahahahahahaha, 2017-03-03:
@azizshahics @xiaolingu    and ,how to plot Power delay profiles ?thanks!

ZhaoZifei, 2017-03-15:
@xiaolingu @azizshahics Have you succeeded in getting CIR (PDP)? I don't know the exact details about this. I will be greatly appreciated if you can share the matlab code with us. Thanks a lot in advance!

ppenumarthi, 2017-04-03:
Hi,
 I somehow don't seem to understand this correctly.
I have recorded CSI values. I am able to get plots directly from CSI values measuring,
Magnitude, Phase of CSI value: 
1. specific to ONE sub-carrier for all packets,
2. Specific to all sub-carriers for ONE packet
I am interested in calculating Power Delay Profile (specific to each packet). Basically, I am trying to replicate this figure. 
[image]

I think need to IFFT to do this. But when I do IFFT, I still receive for 30 different sub-carriers. 

Any help would be appreciated!!

ppenumarthi, 2017-04-11:
Do we get multiple CIR (like CIR for each antenna)? Or only one?I seem to be performing IFFT on CSI at each antenna.

tfsanam, 2017-04-28:
@azizshahics According to your suggestion:

> For time domain, select a subcarrier, say subcarrier # 10, plot packet number on x-axis and corresponding amplitude on y-axis will give you time plot

I got the following plot:
[image]

I have received 693 packets. So here x axis is the number of packet and y axis is corresponding i-th, say 10th subcarieer's csi amplitude. Now my question is, is the x axis is indicationg time in second?
that means, Is it from 0 to 693 seconds?

Please reply at your earliest popssible time.

Thanks.

azizshahics, 2017-04-28:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; }  It is in time domain, x axis indicate the number of packets received.
Regarding total time taken for total received packets you may ask @asif

tfsanam, 2017-04-28:
@azizshahics Thanks. That helps.

azizshahics, 2017-04-28:
whenwhever you figure out on how to identify the time taken in collected all the packets.please share the method os us as well

dindokk, 2017-12-28:
please l need help with how do l plot the data to get the amplitude and phase change

abhinav2938, 2018-11-28:
@tfsanam I am trying to get the similar graph of CSI amplitude vs no. of packets. Could you care to explain how you got this? To be more precise, how did you extract the CSI amplitude value for a single subcarrier from each packet. I am using Matlab and am able to plot the graph of CSI amplitude of different subcarriers for one packet.
Thanks,

mojtabaaghayi, 2019-06-02:
Can sombody send me file csi until i read with matlab 
Tahank
My contact
sm.aghayi@gmail.com

mojtabaaghayi, 2019-06-02:
I start new how do i put in ubentu manually

susan381, 2020-05-15:
@abhinav2938 
Hi, I am trying to plot  CSI amplitude vs no. of packets for each subcarrier. Could you tell me how you solved this problem? I am using Matlab too.
Many thanks,

susan381, 2020-05-15:
@azizshahics 
Hi, could you please tell me how I can plot all the CSI amplitudes of a particular subcarrier  (i.e. subcarrier # 10) versus packet index (or versus time)?
Many thanks

susan381, 2020-05-15:
@azizshahics @abhinav2938 
My data was collected by sending 10 Wi-Fi frames every second for 6 second. Here are more info
[image]

mojtabaaghayi, 2020-05-15:
Hello
yes I can help you
Please send your answer with whats app
My number is +989138523055
I'm wating

azizshahics, 2020-05-15:
+447395404194
  ------------------------------------------------------ Kind Regards,Syed Aziz Shah,
Assistant Professor / Lecturer Room E141, John Dalton Building, Department of Computing and Mathematics, Faculty of Science & Engineering,                       Manchester Metropolitan University.Email: S.Shah@mmu.ac.uk

susan381, 2020-05-15:
@azizshahics  
Any info would be greatly appreciated!

abhinav2938, 2020-05-16:
> @azizshahics
> Hi, could you please tell me how I can plot all the CSI amplitudes of a particular subcarrier (i.e. subcarrier # 10) versus packet index (or versus time)?
> Many thanks

To plot the amplitude of single subcarrier vs no. of packets, what I did was created a loop for getting amplitude of specified subcarrier from each packet received. Took the abs(csi), then added the abs csi i.e. sum(sum(csi,1)). Started another loop from 1 to 56 and took the db value of the sum, ret(i) = db(sum_csi(j)). And plotted the ret, plot(db(ret)).
I did almost a year back and I am not entirely sure of the process.

mojtabaaghayi, 2020-05-16:
Ok
I will send my cod with and you compare me

susan381, 2020-05-16:
@abhinav2938 
Thank you very much for your response! Appreciate it.

mojtabaaghayi, 2020-05-19:
code matlab or python?

mojtabaaghayi, 2020-05-19:
wich is your university?are you student?are you male or famle?