dhalperi/linux-80211n-csitool-supplementary #190: Error in Plot CSI .Dat File

azizshahics, 2016-05-29:
Hello
           I succesfully gathered csi data into .dat file but it shows following message.I would really appreciate if anyone may help
t??? Error using ==> transpose
Transpose on ND array is not defined.

Error in ==> delete at 7
    y=(db(abs(squeeze(csi).')))
 
>> 

thanks in advance.

zainalabidenakhtar, 2016-05-30:
i am also facing the same problem. I have .dat file of 900*1 cell array.

mjer, 2016-06-05:
Hi,
did you check the dimension of csi? 
If you use 3x3 MIMO you'll need to select which transmission antenna you'd like to analyze. In the example code it is assumed that the packet was sent in a single stream and three receiving antennas.

See FAQ Question 2, Paragraph 2 under Nrx
http://dhalperi.github.io/linux-80211n-csitool/faq.html

Hope this helps.
//Martin

zainalabidenakhtar, 2016-06-07:
yest it works if we have 1*3*30 matrix but if we have more number of antennas (more that one Ntx) it gives this error

tahmidzbr, 2016-10-31:
@mjer @zainalabidenakhtar @azizshahics Hi, could you solve the problem?  Ho can i select the transmission antenna that I want to analyze?