dhalperi/linux-80211n-csitool-supplementary #192: Does netlink socket has a buffer for receiving? Log_to_file exit with error: "recv: no buffer space available".

hot-dog, 2016-06-02:
I add a short sleep to while loop, say usleep(5000 * 5) for 25ms and log_to_file exit with error: "recv: no buffer space available" after receive a few packets successfully. I am confused about it, I thought there ought to be a receiving buffer and I use function setsockopt() to set it large enough. Every time when we call function recv() to receive data, it should read all the content in receiving buffer, so recv() should return 252*5 Bytes if the sleep time is 25ms, but it seems not to be like this, it only returns 252 Bytes just like the case I don't add sleep and log_to_file exit with error eventually. Where has the other data gone? Are they just aborted if we don't receive them in time? Sorry for my poor English and I have tried my best. Thx in advance.

hot-dog, 2016-06-06:
The rest data is still in the buffer and they are not discarded. It seems recv() only returns one packet's data(I don't know why it doesn't return all the data in the buffer), so we have to call recv() more than one time to get all the data.

closed 2016-06-06 by hot-dog

zhangshaohu, 2016-06-30:
Do you solve this issue? I tried to set 1000packets/sec,but it will show 'no buffer space available' after running a couple of seconds.

hot-dog, 2016-06-30:
Your issue may be caused by the data rate, it is too high that the program cannot handle. Please refer issue #22 and #33 ,you may find solutions there.

zhangshaohu, 2016-06-30:
Hi, Thank you for your reply. I increased the priority already using command: sudo renice -n -20 -p  pid(log_to_file), I sent 1byte/packet using AP mode. I am not clear about how to add delay time. Is that to modify the log_to_file.c file? where I should put the delay time? could you share me with your modified code? Thank you for your time.

Shaohu

hot-dog, 2016-06-30:
I guess you are not using injection mode, the delay time refers to the time interval between two 
adjacent injected packets and you can specify this delay time. See linux-80211n-csitool-supplementary/injection folder for detail.