dhalperi/linux-80211n-csitool-supplementary #193: Can we convert timestamp_low to the time?

zhangshaohu, 2016-06-06:
I got the timestamp such like:
83719812
83822624
83929756
84028357
84075973
84225927
My question is can we convert timestamp_low to the data/time? If not, how can I record the time? any suggessions? Thanks!

closed 2016-06-29 by zhangshaohu

ogutiann, 2016-11-01:
hello, can you help to explain for me this if you figured it out? i would like to plot Amplitude Vs Time but i am unable to figure out the time axis.

zhangshaohu, 2016-11-01:
What I did is to modify the code in log function in CSI tool.  I added one line code to get the the linux time which was saved in a separated file , e.g. csv file. In short, you will log one line time  when you log one line csi data.  I combined the time with the csi value after I used matlab to extract csi.
It looks like:
7:27:50	456632625
7:27:55	456635574
7:28:00	461634464
7:28:09	461664993
7:28:14	466637372

Remember, the timestamp in CSI tool repeat again and again, it is not the real time .

swashah, 2016-11-20:
@zhangshaohu : Is there any way to synchronize data on low laptops by using time stamp? Did you perform any thing like that or have any idea?