dhalperi/linux-80211n-csitool-supplementary #194: Questions about the recorded timestamps.

HSID, 2016-06-07:
From two experiment trials, I got two kinds of timestamps.

One is in this kind of format "537739414".
The other is like "1.844674407225976e+19" .

How to explain these types of timestamps? How to convert them into number of time units?

Thank you!

vinatby, 2016-06-07:
You need to consider data rate and number of samples you have received. then convert it to minutes by dividing it by 60. 

divide it by data rate ( you can find that it in struct), and this data rate comes from how often you ping your AP.

I guess this numbers are being counted since your router was first in operation. If you plot the graph of this time-stamps from all samples; you will get almost linear graph (some data packet loss).

Correct me if I am wrong at some point. I was working on this project in past and then dropped. otherwise I hope this helps

caichao, 2016-06-17:
@HSID I also have some puzzles about the time stamps. When I ping the router using 1 Hz rate, the time difference between adjacent samples varies. I divide the time stamps by 10^6, sometimes, it gets the true value, but sometimes the output is like this:

time difference  : 

ans =

   1.4000e-04

time difference : 

ans =

   3.5100e-04
I wonder if the time stamps can be useful to indicate the sampling rate.

PigSea, 2016-06-22:
@caichao since timestamp_low is the low 32 bits of the NIC's 1 MHz clock. It wraps about every 4300 seconds, or 72 minutes. I think the time stamps circule every 4300s. so if the recorded timestamp does't pass 4300s, we can divide the time difference by 10^6 to get the lasting time, otherwise ,  it will be a negative number. Correct me if I am wrong.