dhalperi/linux-80211n-csitool-supplementary #195: About the frequency spacing between two consecutive subcarriers

shaozhuchuang, 2016-06-08:
@dhalperi @dpward if i use HT20, what is the frequency spacing between two consecutive subcarriers? 20 MHz/64 or 20 MHz/56 or 20 MHz/30? Thank you!

asifhanif1122, 2016-06-08:
@shaozhuchuang Frequency spacing will be 20MHz/64 = 312.5 kHz. 
For 40MHz bandwidth number of sub-carriers increases to 128. Therefore, HT40 will also result in same frequency spacing 40MHz/128=312.5kHz.

shaozhuchuang, 2016-06-08:
@asifhanif1122  From FAQ,I can know that CSI Tools can only measure 30 subcarriers. why divide 64 and 128?   Thank you!

asifhanif1122, 2016-06-08:
Only 30 sub-carriers are reported. That is correct but it does not mean the number of  OFDM sub-carriers has decreased. In 802.11n standard 64 sub-carriers are used for 20MHz wide band channel.

shaozhuchuang, 2016-06-08:
@asifhanif1122  what the 128 represent?

asifhanif1122, 2016-06-08:
128 is the number of OFDM sub-carriers in a wifi channel of 40MHz bandwidth.

asifhanif1122, 2016-06-09:
@shaozhuchuang When we use 20MHz wide channel, we split 20MHz bandwidth among 64 OFDM sub-carriers and for 40MHz wide channel , bandwidth is divided across 128 sub-carriers. In both cases spacing between two adjacent sub-carriers will remain same i.e 312.5kHz. Table you posted above shows grouping of sub-carriers to report CSI. In grouping only selected sub-carriers are used to report CSI so that feedback frame size may not become too large. CSI values for  missing sub-carriers are extracted by using interpolation. Sir D.Halperi modified firmware that reports CSI only for 30 sub-carriers out of 64(for 20MHz wide channel) or 128(for 40MHz wide channel) sub-carriers.

asifhanif1122, 2016-06-09:
http://pdcc.ntu.edu.sg/wands/yxxie/papers/Mobicom15_Recitation.pdf

Please read section 2.1 of this paper for further understanding. They have not used Intel 5300 NIC but basic concept is same.

zainalabidenakhtar, 2016-06-14:
@asifhanif1122 how we can come to know which frequency (20MHz or 40MHz) we are using? can you please share command to know it.

asifhanif1122, 2016-06-14:
@zainalabidenakhtar I don't know about command to confirm bandwidth of wi-fi channel. But selected channel can easily be confirmed using _**iwlist channel**_ command. At the bottom of output of this command you will be able to view your currently selected wi-fi channel.

zainalabidenakhtar, 2016-06-15:
thanks @asifhanif1122 i try this

azizshahics, 2016-07-31:
@asifhanif1122  Can you please indicate how to plot phase response of CSI we obtain?