dhalperi/linux-80211n-csitool-supplementary #199: My csi samples can not be fed into SpotFi algoritm

caichao, 2016-06-17:
I am trying to sample my own csi data and validate the spotFi algorithm. https://github.com/egaebel/lgtm/tree/master/csi-code
This is the code I use. But when I run the code using my own csi data samples, only part of samples can output the estimated aoa and tof. Lots of the samples will output the same error like this:
Full Measurement Matrix Size: 0
Subscripted assignment dimension mismatch.

Error in spotfi (line 144)
    full_measurement_matrix(:, 1) = full_measurement_matrix(:, 1) / aoa_max;

Error in spotfi_analysis (line 15)
spotfi(matrix,carrier_frequency,channel_space,antenna_space); 

All the samples are collected in an indoor environment. The NIC5300 works in client mode, with each antenna spacing 0.02m. The AP has 2 antennas. And I have also tuned different sampling rate but still it didn't work. 
I am not sure whether this problem is caused by the weighted values in the likelihood estimation part.
Can anybody offer a help? Thanks in advance.

caichao, 2016-06-17:
It seems that the problem is caused by function aoa_tof_music, the output of this function is 
K>> aoa_packet_data{1}

ans =

   Empty matrix: 0-by-1

caichao, 2016-06-17:
The plotted Pmusic is in the following figure:

[image]

I wonder if this problem is caused by my hardware imperfection.

ozawali, 2016-06-22:
Spotfi is right you are wrong!

caichao, 2016-06-22:
@ozawali Thanks for your reply. I know I was wrong. But I just couldn't figure out the problem. Can you help me?

asifhanif1122, 2016-06-22:
@ozawali Can you please look into my problem. When i get CSI data in file I pass it through **_get_scaled_csi()_** function and plot SNR then values come out to be negative. Which probable thing am I while plotting SNR? 

My Matlab code snippet 

csi = read_bf_file('myfile.dat');
output= get_scaled_csi(csi) ;
plot(db(abs(squeeze(output).')))
legend('Antenna-1', 'Antenna-2', 'Antenna-3');
xlabel('Sub-carrier Index');
ylabel('SNR (dB)'); 
[image]

Poomathi, 2016-07-14:
@asifhanif1122  In the link for SPotfi program (https://github.com/egaebel/lgtm/tree/master/csi-code) I couldn't find the program for "get_scaled_csi" function and also inside  read_bf_file_modified() file, function read_bfee() function is missing. Please correct me if I am missing something here.

YuexinZhao, 2016-07-17:
@caichao Have you solved the problem that there is no peaks in the plotted Pmusic? I think that there     is a bug in Sanitization Algorithm for SPotFi program [https://github.com/egaebel/lgtm/tree/master/csi-code](url) .

Poomathi, 2016-07-20:
I used Spotfi to run the simulated data using collectplaneWave matlab comment (2 multipaths and one direct signal) without any noise, Program is not working as expected. For example if I use -10 azimuth for direct path, SpotFi estimates it as -56. If the same data is used in Matlab inbuilt command( phased.RootMUSICEstimator), it predicts the azimuth as expected as -9.98. Program looks correct to me but something is going wrong here.

Poomathi, 2016-07-20:
I meant SpotFi code at this below location
https://github.com/egaebel/lgtm/tree/master/csi-code

Poomathi, 2016-07-21:
https://github.com/egaebel/lgtm/tree/master/csi-code  In Spotfi program changing spectrum calculation from "PP = steering_vector' * (eigenvectors * eigenvectors') * steering_vector;" to "PP = sum(abs(steering_vector.' * eigenvectors).^2,2);" provides correct azimuth as expected. transpose ' earlier not only reverses the columns and rows but it also take conjugate of the complex values. so changing to .' solved the problem. Now I see peaks at correct location. Thanks to "egaebel" for sharing the code.

caichao, 2016-07-22:
@YuexinZhao sorry, I didn't solve the problem right now. I think the antenna space matters so I will try to change the hardware of the WiFi adapter. 

@Poomathi Thanks for sharing your experience.

YuexinZhao, 2016-07-22:
@caichao Thanks for you reply.Could you leave a mailbox for discussion in details?

caichao, 2016-07-22:
@YuexinZhao caichao08@gmail.com

soltanaghaei, 2016-07-27:
dear @caichao and @YuexinZhao, I'm also working on SpotFi paper and thing what you discussed about the PP calculation is wrong, as in the SpotFi and ArrayTrack paper, they both used the notation "H" or "*" which is the conjugate transpose or Hermitian transpose of a matrix. So, I think using the transpose ' is correct and there should be another mistake in the code to generate wrong AoA spectrum. Did any of you find the bug by any chance?

caichao, 2016-07-29:
I am still trying....