dhalperi/linux-80211n-csitool-supplementary #202: How can I change the txpower in injection mode?

HSID, 2016-06-23:
I use the command 'iw mon0 set txpower fixed 10000' to set the txpower of my transmitter and the command 'iwlist txpower' to check out whether or not the txpower is successfully changed.
It seems that the txpower is always fixed at 15dBm.
What is the problem? How can I change the txpower?
@dhalperi @dpward

DontKnowMy, 2017-03-31:
I would like to know the same. Has nobody an answer?

Thanks!

lqh929289158, 2017-04-07:
l have the same question.
I have tried many the possible solutions provided on the internet.
`sudo iwconfig wlan0 txpower 20`
then use the `iwconifg` to check it. NO changes.
`sudo iw reg set BO`
`sudo ifconfig wlan0 txpower 20`
then use the `iwconfig` to check it. NO changes.
I just want to know how could I make the txpower lower?

asifhanif1122, 2017-04-07:
I think @dhalperi might have disabled tx power level modification. Considering the procedure of CSI calculation, the power level of transmitted signal must be known prior CSI calculation(my speculation). So that may a reason behind fixed power level(15dBm) in Intel 5300 NIC after firmware modification.