dhalperi/linux-80211n-csitool-supplementary #203: about CSI data

ozawali, 2016-06-27:
@dpward @dhalperi   I want to add the mac address to the CSI data belongs to the sender So if I need modfied the kernel for the iwwifil?
Thanks a lot

hamaoui, 2016-06-30:
i'm in the same boat, i want to have CSI data for each mac address. Any help ?

skocharllge, 2017-07-16:
@hamaoui @dhalperi @dpward 
@ozawali 
Could you please tell if you were able to figure out this problem.I want to specifically process CSI data from 2 nodes(as of now).The way to specifically distinguish that data would be to have a mac address as part of CSI.

The MPDU data provides a 802.11 mac header but that comes independently to the application layer which is with tag 0XC1.Beamforming data comes with 0XBB.How does the application know which beamforming data pertains to a mac header?