dhalperi/linux-80211n-csitool-supplementary #209: number of samples per second

zainalabidenakhtar, 2016-07-13:
hello! can any body guide me how to set number of samples per second? i am using managed mode at 2.4GHz frequency band. for example if I want to send 2500 sample per second, how to set this?

asifhanif1122, 2016-07-13:
[image]

asifhanif1122, 2016-07-13:
[image]

asifhanif1122, 2016-07-13:
You can find number of sample per second by dividing number of packets with total time taken by packets at transmit side.

zainalabidenakhtar, 2016-07-14:
@asifhanif1122 thanks for your response, when I try this i get a message " sudo:  ./random_packets: command not found"

chendachong, 2016-09-07:
Hi, I meet the problem too.Now, do you have solved the problem,if so,would you please guide me how to get more samples per second?

tahmidzbr, 2016-10-03:
@chendachong @zainalabidenakhtar @asifhanif1122 
hi im having the same problem:
sudo: ./random_packets command not found.
please help.