dhalperi/linux-80211n-csitool-supplementary #210: Problems Connecting with a Mobile Phone in AP mode

Carll7, 2016-07-14:
I have a similar problem as in #65.
Namely, when the computer boots up the OS is loaded with
`[   11.157636] iwlwifi 0000:02:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm` 
However, when I connect the mobile phone there is a change in loaded firmware/driver
`[  331.696011] iwlwifi 0000:02:00.0: Loaded firmware version: 8.24.2.12`

Repeating the steps in http://dhalperi.github.io/linux-80211n-csitool/installation.html does not help.
I am unable to connect to AP after that. There is no such problem when connecting from a laptop.

Using the dmesg | grep -i wifi shows I noticed

`[  331.695220] iwlwifi 0000:02:00.0: Microcode SW error detected.  Restarting 0x82000000.`

among other output. Is this the firmware/driver problem???

skocharl, 2017-07-24:
@dpward @dhalperi @rakeshbala 
Could you please share on below issue ?
The Intel5300 comes up on AP mode properly but when my iPhone is connected to AP it keeps disconnecting and I can see the logs on hostapd as "disassociated" "deauthenticated due to inactivity (timer DEAUTH/REMOVE)

pcanwar, 2018-05-14:
check this tool
https://github.com/pcanwar/Connect-smartphone-to-the-CSI-tool