dhalperi/linux-80211n-csitool-supplementary #213: the structure of csi_entry

haimapi, 2016-07-18:
### hello, dear all:
I have 2 Tx antennas and 2 Rx antennas, so the csi should be ：csi: [2x3x30 double]
however sometimes, with the condition that the number of Tx and Rx is the same, I got：csi: [1x3x30 double]. Do anyone can help with this? How can this happens? Thanks a lot!

swashah, 2016-11-06:
@haimapi : CSI should be [2x2x30] not [2x3x30]

tahmidzbr, 2016-11-09:
@haimapi @swashah Did you figure out why sometimes the CSI is measured as 1x3x30 and sometimes as 2x3x30?

swashah, 2016-11-09:
@tahmidzbr : it is due to the reason that transmitter is adapting its own rate selection...

tahmidzbr, 2016-11-10:
@swashah Thank you. But i see that recently my transmit antennas are always fixed at 2. But my receiver which has the Intel 5300 NIC the antenna C drops every now and then, i.e. no RSSI at Rx:C. Do you know why so? Also in the CSI matrix, the rows represent Nt and columns represent Nr, but how is the Nr column arranged? Is it always 1 2 3 or does it depend on the perm value?

swashah, 2016-11-10:
CSI matrix is arranged in order 1 2 3 for both Nt and Nr. I have seen some where that for this NIC the central antenna (i.e. antenna C ) is normally dropped and I guess it is due to its position.

haimapi, 2016-11-10:
@swashah @tahmidzbr ,thank you and yes, this is extractly what i confront, 1x3x30 and 2x3x30, why?

tahmidzbr, 2016-11-10:
@swashah So antenna C is the central antenna?
Is there any link i can take a look how it is actually arranged? So basically A=1 B=3 and C=2? is this the correctway of antenna indexing?
@haimapi  actually i dont have the issue of my transmit antenna being changed, its always the received antenna number that changes,..sometimes Nr=3 and sometimes Nr=2, But Nt=1 always. 
Does anyone know how i could fix the transmit power of both my transmitter (router) and the receiver (Intel NIC 5300)?