dhalperi/linux-80211n-csitool-supplementary #217: What is Effective SNR?

ajebulon, 2016-07-27:
I found in the FAQ link mentioning about "Effective SNR". I already looked for it in Google but I could not find one explanation matching with the "Effective SNR" mentioned in the FAQ link. There is no explanation about it either in the FAQ.

Could someone please explain to me what is "Effective SNR"?
@dpward @dhalperi 

Thank you.

tahmidzbr, 2016-10-03:
@ajebulon Hi could you figure out the meaning? if so can you please explain? how can i relate CSI with SNR?

ajebulon, 2016-10-04:
@tahmidzbr I haven't found any answers yet, so sorry, I can't help you.

2001dandelion, 2020-03-31:
In multi-carrier system, where a user can simultaneously use different carriers to transmit its data, each carrier will have a different instantaneous SNR.  To calculate the link rate (the sum rate of the used carriers) or outage probability, we have to consider all the instantaneous SNR of used carrier in the calculating process, which is time consuming. Thus, to simplify the calculating process, we have a term called effective SNR, which can be used to replace the multiple carrier's SNR in the calculating process. 

Reference: "Exponential Effective SINR Approximations for OFDM/OFDMA-Based Cellular System Planning"

closed 2020-03-31 by ajebulon