dhalperi/linux-80211n-csitool-supplementary #218: What is the carrier frequency in 40 Mhz channels?

soltanaghaei, 2016-07-27:
Does anybody knows what should be the central frequency when we use 40Mhz bandwidth with HT40-? I've found this information by reading previous asked questions:

> "40 MHz channels are defined by pairing two 20MHz channels: a control channel and an extension channel. When we say "64 HT40-" that is two separate arguments. The first says "use channel 64 as the control channel". The second says "use a 40 MHz channel with the extension channel below the control channel". This means, therefore, the 40 MHz channel consisting of {64, 60}.

but it made more confusing in terms of the carrier frequency? should I still consider the frequency of control channel as the carrier frequency? 

@dhalperi @dhalperi @linkin8834 @ozawali @asifhanif1122

linkin8834, 2016-07-27:
Hi,
    You can try iw dev wlan0 or something like that. It would show you the center frequency. I calculate the center frequency as the middle of the lower channel and upper channel. e.g., channel 48, HT40-, then the center should be the middle of channel 44 and channel 48.

soltanaghaei, 2016-07-27:
Dear @linkin8834 Thank you for your prompt reply. However, when I run command " sudo iw wlan2 set channel 48 HT40-", and then check the interface information with iwconfig, the frequency is 5.24GHZ which equals to channel 48!! Are you seeing the different value in your case?

linkin8834, 2016-07-27:
@soltanaghaei 
    iwconfig show you the main channel center frequency I think. You should try iw dev wlan0 or something like that.

soltanaghaei, 2016-07-27:
@linkin8834  "iw dev wlan0" does not have parameter to show the frequency. In addition, as I'm working in injection mode, the interface is not connected to any access point. I've searched a lot and couldn't find any command to give me the center frequency. Do you know any or have your used any?

linkin8834, 2016-07-27:
@soltanaghaei You can try to configure one of your laptop as an AP using hostapd, and then type 'iw dev' in the command window. I can see the center frequency when I type iw dev on my software AP. Anyway, I figure out that the center frequency under 40MHz configuration is the middle of the two 20MHz channels. For instance, I saw this:

channel 40 (5200 MHz), width: 40 MHz, center1: 5190 MHz

soltanaghaei, 2016-07-27:
Great, thank you for your helps :)