dhalperi/linux-80211n-csitool-supplementary #221: Which tx antenna is used for transmission in injection mode?

soltanaghaei, 2016-07-29:
Does anybody know how I can assign or find which tx antenna is used to transmit packets in 4101 rate in injection mode (which is the rate for one stream)? Is it possible that the selected tx antenna will be changed during transmission or every time? How can I check that? 

Thank you 

@dpward @dhalperi @linkin8834 @ozawali @asifhanif1122 @Boeing738ng

linkin8834, 2016-07-29:
I think the 4101 rate is associated with either Antenna A, B, or C, corresponding to port 1, 2, and 3 on your WiFi card (or port 0,1, and 2). Anyway, you can use a cable to verify which TX is actually transmitting signals.

If you are using two 5300 laptops with injection/monitoring, I think that the TX antenna would change. It  is fixed and dependent on your rate setting, e.g., 4101. If you are using a router with a 5300, then I am not sure if the router would use rate adaption to alter its transmission antennas and switch between different MIMO configurations.

dhalperi, 2016-07-29:
Please read FAQ number 1. https://dhalperi.github.io/linux-80211n-csitool/faq.html

closed 2016-07-29 by dhalperi

lidongning, 2017-06-19:
you mean in  injection/monitoring,   if I set rate , TX won't  be changed,  and TX- stream is fixed???  @linkin8834

ozawali, 2017-07-18:
@dpward @linkin8834 the paratemtes used for changing Tx antenna maybe disable in injection or monitor mode.