dhalperi/linux-80211n-csitool-supplementary #222: can not log payload info to file using injection/monitor mode

liucc07, 2016-07-30:
If I make the driver works in station mode and connect it to an AP with "connector_log=0x5" based on FAQ.12. It can successfully log both the payload and CSI into the file. 

However, if i set it work in monitor mode and another in inject mode, only the CSI is loged into the file. 

Can anyone help me figure it out?

liucc07, 2016-07-30:
@dpward @dhalperi @ippeb

closed 2016-08-03 by liucc07

swashah, 2016-11-22:
@liucc07 : will you mind sharing that how to log payload information? what modifications did you make in code. and also does setting 'connector_log=0x5' gives any packet sequence number or not?

soltanaghaei, 2017-01-23:
@liucc07 I have the same issue and I can not get the payload in any case with monitor/inject mode. Could you finally figure out any solution for this issue?