dhalperi/linux-80211n-csitool-supplementary #223: Can‘t connect to the access point

chendachong, 2016-07-31:
My notebook has only 2 antennas, and I use the Intel 5300AGN to get CSI. But after I followed the steps mentioned in the Installation Instructions, I can't connect to an 802.11n access point. Who knows the reason.

TadiGuy, 2016-07-31:
Were you able to join the AP before installing CSI tool?
Can you see the SSID for the AP you are trying to join from your laptop?

chendachong, 2016-08-01:
Now, I can connect to the access point  with no  password. What can I do to make it connect to any access point  with  password

tahmidzbr, 2016-09-09:
Do you know why you couldnt connect before but now you can? I cant connect to an access point that has no password.

And to your question, i think the code was written to only connect to unencrypted APs