dhalperi/linux-80211n-csitool-supplementary #226: How to find phase response of CSI data received

azizshahics, 2016-08-01:
HI
      I would like to analyse the CSI measured data into phase response.
I would really appreciate if anyone can help.

regards

sheheryar9, 2016-08-01:
Hi, 
    I am using an AP and a receiver which is a laptop equipped with Intel5300. My AP has 3 transmit antennas and my receiver has 3 receive antennas. So that means that my CSI matrix is 3x3x30. I always get the values for channel link between 1 and 1. But whenever I try to use 1x2 , 1x3 , 2x2 or any other combination except 1x1 I always get error that the vector has some values which are NaN so can anyone of you tell me what is the reason for this ? I would really appreciate if anyone can help. Thanks. 

Best Regards,
Sheheryar

muneebaraja, 2017-05-16:
@azizshahics did you find a way to find the phase shift. I have only 1 transmitter and receiver.