dhalperi/linux-80211n-csitool-supplementary #228: csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6');

chendachong, 2016-08-03:
There is something wrong, shows"Command not found"

tahmidzbr, 2016-10-03:
@chendachong did you figure it out yet? if so can you please tell me how you did it?

ogutiann, 2016-10-19:
@chendachong and @tahmidzbr, you should process CSI using matlab or octave. running the commands: csi_trace = read_bf_file('sample_data/log.all_csi.6.7.6'); in Ubuntu command line will not work.

tahmidzbr, 2016-10-19:
@ogutiann I figured my problem is log_to_file csi.dat is not working. Is it working for you? Can you please help me?
The problem is when i run the command sudo ...../netlink/log_to_file csi.dat and then open a new terminal to ping to my AP, i see that the ping is correct (i.e.receive replies from AP) but nothing gets recorded in my log_to_file csi.dat

can you please tell me what you did?

I checked my other functionalities:
*modified firmware is installed correctly
*using unencrypted AP

ogutiann, 2016-10-20:
@tahmidzbr to get everything working, i followed these two explanations, #234 and #45. make sure that before you ping your AP, run this command sudo dhclient wlan0.

rt6n1, 2018-04-15:
I experience the same problem as @tahmidzb. sudo dhclient wlan2 (my wlan adapter) did not help.