dhalperi/linux-80211n-csitool-supplementary #229: Can not set channel to HT40+ in injection mode!

soltanaghaei, 2016-08-03:
Did anybody have the issue to set the channel to  HT40+ in injection mode? I can run the commands for HT20 and HT40- without any error, but for HT40+, I received "command failed: invalid argument (22)". Can you guess what the issue is? 

@dhalperi @dhalperi @linkin8834 @ozawali @asifhanif1122

asifhanif1122, 2016-08-04:
Which channel are you using?

soltanaghaei, 2016-08-04:
I'm using channel 48 in 5GHz.

sent from Gmail on Android

asifhanif1122, 2016-08-04:
Apparently there should not be an issue if channel 48 is being set as primary channel. Whenever huge jump is present in next available  channel, HT40+ does not work. Like if you use HT40+ with channel 64 , then it will not work as next available channel is 100. 
See channel list here.
[image]