dhalperi/linux-80211n-csitool-supplementary #230: What is the effect of HT on channel response?

soltanaghaei, 2016-08-09:
Hi all, 
Can anybody justify the reason of seeing different SNR values for HT20, HT40- and HT40+? This is the output of my experiments in these three situations:
[image]

The experiments are performed in exactly similar situations (empty room without replacing tx and rx computers- injection mode- same transmition rate). I see the same trend in different experiments: (1) SNR drop on both end sides of subcarrier range in HT20, (2) SNR drops on smaller subcarriers for HT40+ (3) and drops at the end of subcarrier range in HT40-. Is there any specific characteristic in NICs that causes this behavior? 

In addition, I expected to see similar results for CH 44 HT40+ and CH48 HT40- as the channels and subcarriers should be similar,, but the results are completely different, why? Are the subcarrier frequencies different in these two scenarios? 

@dhalperi @dpward  @linkin8834 @ozawali @asifhanif1122 @ippeb

soltanaghaei, 2016-08-09:
Dear  @linkin8834 do you have any suggestion for this question as well? 
" I expected to see similar results for CH 44 HT40+ and CH48 HT40- as the channels and subcarriers should be similar,, but the results are completely different, why?"

tahmidzbr, 2016-09-27:
Hi @soltanaghaei  could you figure out why? Also can we choose HT40+/HT40- when running our simulations? If so can you please tell me how? If i also do it then there is a chance we two can collaborate on this idea and find out the reason for the difference in SNR