dhalperi/linux-80211n-csitool-supplementary #231: Jumps in relative phase values

soltanaghaei, 2016-08-09:
Hi All, 
Does anybody know why there are some jumps in relative phase values? I can guess that these jumps are 2pi apart and should be almost the same phase values, but Why does it happen only for some specific packets? 

[image]

@dhalperi @dpward @linkin8834 @ozawali @asifhanif1122 @ippeb

linkin8834, 2016-08-09:
Low amplitude on edges: result of filtering
Sudden jumps: do a phase unwrapping and check again.

> Hi All,
> Does anybody know why there are some jumps in relative phase values? I can
> guess that these jumps are 2pi apart and should be almost the same phase
> values, but Why does it happen only for [...]

soltanaghaei, 2016-08-09:
Thanks @linkin8834 for your reply. Can you please introduce me a reference to read about filtering effect? Why is it different in HT20 and HT40?

I agree that phase unwrapping may solve the issue, but why should it happen for some packets only? does this 2pi shift show anything?

linkin8834, 2016-08-09:
Phases are all wrapped into pi and -pi. Imagine a linear phase growth
starting from -pi and barely touch +pi at the end. Without phase noise
there is no phase jump. But unforfunately there is. Whenever a point on the
straight line touches -pi or +pi due to noise you see a sudden jump.

Generate exp(-i2pik) with k from -1/2+eps to 1/2-eps. Then add a noise onto
it and plot the angle.

For the filtering, check rrc filter. I also saw this on usrp csi
estimations and believe that it is due to filtering

CrystalWong2, 2016-09-03:
@soltanaghaei I did similar experiment too. I want to know how to plot the CSI phase. Could you please help me? Or could give me some reference which could help me understand it.

tahmidzbr, 2016-11-16:
@CrystalWong2 @soltanaghaei @linkin8834 
Hi all, I tried to plot the phase, but in my case my phase jumps are not from -pi to +pi. Yes the jumps are always in between them, but they dont reach -pi to +pi. Can anyone please suggest why?

DneprovV, 2017-05-31:
Hello, @soltanaghaei . I know the time has passed, but maybe you are still here. I'm new to this theme and I took yours picture as reference. Can you please provide some info about your first figure. How many csi packets do you have there (length(csi_trase)), what are experiment conditions (distance, LOS or NLOS, 2.4/5 GHz and smth like this)? I'm confused with my results on picture below. I have there 638 csi packets, and I dont like stability of relative phase and SNR. I need to figure out if I'have made mistake somewhere or my channel is unstable. Can you provide .m script that plots your figure? 

[image]