dhalperi/linux-80211n-csitool-supplementary #232: How to display the csi values in real time?

chendachong, 2016-08-10:
Is there any software to display the csi values in real time? I want to see the changes in real time, not use the matlab.

libing64, 2016-08-11:
You can try rviz.

zhangshaohu, 2016-08-18:
Bing, I want to show my project in real time as well. Could you give more detail about how do you use the rviz to present the csi value in real time. Thanks a million. 
Shaohu

tahmidzbr, 2016-10-30:
@chendachong @libing64 @zhangshaohu 
Did anyone figure out how to capture the packets in real time?

mustafasadiq, 2016-11-13:
Hello everyone
Does anyone knows how to draw the CSI data in matlab in real time while capturing the CSI data?

lubingxian, 2017-04-10:
Hi, @chendachong @zhangshaohu @tahmidzbr @mustafasadiq 
Here is my solution. https://github.com/lubingxian/Realtime-processing-for-csitool