dhalperi/linux-80211n-csitool-supplementary #233: Is the order of rx columns in CSI matrix based on NIC labels or "perm" attribute?

soltanaghaei, 2016-08-11:
Hi, 
Does anybody know if the rx columns of CSI matrix is in the same order as the antenna numbers marked on the NIC or is based on "perm" attribute? I've noticed that the read_bf_file() function takes the perm attribute into account as follows:
` ret{count}.csi(:,perm(1:Nrx),:) = ret{count}.csi(:,1:Nrx,:);
`
Does it undo the permutation so that the CSI matrix matches with the antenna numbers? 

@dhalperi @dpward @linkin8834 @ozawali @asifhanif1122

balazsbme, 2016-08-18:
I am also interested in the answer for this one! :)

tahmidzbr, 2016-11-10:
@soltanaghaei Very good question. Did you figure it out? If so can you please share?

ozawali, 2017-03-02:
the Matix has the right sequence

ozawali, 2017-03-02:
the matrix has the right sequence

PigSea, 2019-06-25:
see #112. After read_bf_file(), the csi matrix is in the right order: A, B and c

PigSea, 2019-06-25:
In read_bg_file.m, "ret{count}.csi(:,perm(1:Nrx),:) = ret{count}.csi(:,1:Nrx,:);" undo the permutation based on "perm" attribute.