dhalperi/linux-80211n-csitool-supplementary #234: The wifi networks disappear after unloading the driver

siamaky1984, 2016-08-16:
Hello,
I have bought a 5300 Intel NIC and have replaced with the old NIC.
I have installed Ubuntu 12.04 and have implemented the code given on the website.
The process goes fine until the end where I unload the driver and reload it which causes the wireless networks to disappear.
I can't understand how I can get connected to a specific WiFi AP from here based on the information you have provided. How should I use iw and iproute2 utilities?
I highly appreciate your response in advance.
Sia

zhangshaohu, 2016-08-18:
If your installation is correct, you may use these command:

make -C linux-80211n-csitool-supplementary/netlink
sudo modprobe -r iwlwifi mac80211
sudo modprobe iwlwifi connector_log=0x1
sudo ifconfig wlan0 up
sudo ifconfig eth0 up
sudo iwconfig wlan0 essid TP-LINK_DD9C      //link your router name
sudo ls /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/
[sudo] password for wens: 
bc:46:99:d6:dd:9c

sudo cat /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/bc:46:99:d6:dd:9c/rate_scale_table
sta_id 0

echo 0x1c90e | sudo tee /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/bc:46:99:d6:dd:9c/rate_scale_table
sudo iwconfig wlan0 essid TP-LINK_DD9C
sudo dhclient wlan0
sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat

//open another terminal 

pgrep log_to_file
sudo renice -19 -p pid(pid of log_to_file)

ping -i 0.05 192.168.1.1     // ping your AP address

Hope this helps.

zhangshaohu, 2016-08-18:
If you still have some error, you can try  to unload and load driver module  many times util see the WiFi light is on.
or 
sudo ifconfig wlan0 down
sudo ifconfig wlan0 up

duanabao, 2016-08-18:
I also encounter the same problem issued by @siamaky1984 .  I used the ubuntu 14.4.1(32 bits,kernel version is 3.13.0-32.57 ). The network disappeared when I unload the driver and reload it. When I use ip link set wlan0 up, the wlan0 can not be in on state.

I used the commands provided by @zhangshaohu . But I cannot find any file in the directory /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations . the 5300 NIC cannot scan any wireless ap.

I wonder if the driver is not built rightly, even if no problems exist during the building process. Thanks for any help.

siamaky1984, 2016-08-18:
Hey Shaohu,

Thanks a lot for your response but I have issues running your code. The
place where you use:
sudo/ls/sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/
[sudo] password for wens:
bc:46:99:d6:dd:9c

What version of Ubuntu are you using by the way?

zhangshaohu, 2016-08-18:
@siamaky1984 
First of all,  you have to make sure the wlan0 is on, and then
 `sudo ls /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/`  
type your password, it will show the station address.

I tried many versions. I found this Ubuntu version works for me.
Download website:
http://old-releases.ubuntu.com/releases/14.04.2/
version:Ubuntu-14.04.1-desktop-amd64.iso
 22-Jul-2014 22:36  1.0G  Desktop image for 64-bit PC (AMD64) computers (standard download)

The kernel tag is 
UBUNTU_KERNEL_TAG=Ubuntu-3.13.0-32.57
I create a repository in my Github, I will update my project once I am available
[https://github.com/zhangshaohu/WiT](url)

zhangshaohu, 2016-08-18:
@duanabao
Make sure you installed CSI tools correctly.
How about the output using the command `iwconfig`.
In my case:
wens@wens:~$iwconfig
eth0      no wireless extensions.

lo        no wireless extensions.
wlan0     IEEE 802.11abgn  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=15 dBm   
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Power Management:off       
wens@wens:~$ iw dev wlan0 link
Not connected.
wens@wens:~$ iw wlan0 connect TP-LINK_DD9C
command failed: Operation not permitted (-1)
wens@wens:~$ sudo ifconfig wlan0 up
wens@wens:~$ sudo ifconfig eth0 up
wens@wens:~$ sudo iwconfig wlan0 essid TP-LINK_DD9C
TP-LINK_DD9C is my AP name

duanabao, 2016-08-18:
@zhangshaohu Thanks very much for you reply.
1) $iwconfig, I can see the following information:
   wlan0 IEEE 802.11abgn ESSID:"TP_LINK_A73B76"
     Mode:Managed  Access Point:Not-Assocaited  Tx-Power=off
     Retry long limit:7    RTS thr:off Fragment thr:off
     Encryption key: off
     Power Management:off
  
lo:  no wireless extensions.
eth0: no wireless extensions.

2)  $ sudo ls /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/
  I just see nothing after the said command. my computer cannot connect with AP.

Can you give me more suggestion?

Thanks

siamaky1984, 2016-08-18:
Actually I could connect to an unencrypted AP I had using : $ iw dev wlan0 connect {My SSID}
But now when I run: 
$ sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat
and ping {MY IP} on another Terminal, I do not see any change in csi.dat file.
Anyone has faced similar issue?

zhangshaohu, 2016-08-18:
@siamaky1984 
Have you run this command before logging CSI ?
`sudo dhclient wlan0`
Please carefully check your command step by step.
If you installed CSI tools correctly and can load modified driver. It should work.

zhangshaohu, 2016-08-18:
@duanabao 
You can see the EESID.  I think you have already connected the AP, but you should turn on the eth0

duanabao, 2016-08-19:
@zhangshaohu  Thanks a lot.
1) my AP require passwords to connect. So when I use command "iwconfig", I can see the EESID, but in fact, my computer has not connected to the AP, it just connect and disconnect.

2) I cannot get ip address also, when I used the command "ifconfig wlan0".

I also used commands "ip link set eth0 up".  

3) when I use wpa_supplicant to connect the encrypted AP, I got the error message.
wlan0: trying to asscocaite with 5c:63:bf:a7:3b:76
ioctl[SIOCSIWFREQ]: Device or resource busy
wlan0: Association request to the driver failed
wlan0: CTRL_EVENT_DISCONNECTED bssid=5c:63:bf:a7:3b:76 reason=0

Any suggestion? Thanks a lot

siamaky1984, 2016-08-19:
@zhangshaohu, That's right. I also used:
sudo dhclient wlan0
I can't figure out what the problem is.

zhangshaohu, 2016-08-19:
@duanabao Note that you should use the AP without encryption. Simply, you don't have to connect the router with the internet cable.

niuqun, 2016-09-23:
@zhangshaohu Hello, Mr. Zhang! I ran into similar problems with @duanabao. I have two questions regarding your reply.
1) How do you get 0x1c90e in "echo 0x1c90e | sudo tee"?
2) Why do we need eth0 at all?
Thank you!

tahmidzbr, 2016-10-17:
@siamaky1984 Hi could you figure out the problem? I also cant log_to_file csi.dat as I can see my csi.dat has 0 bytes! Can someone please explain why so?
I did everything @zhangshaohu  said and i could ping and transmit and receive packets

ogutiann, 2016-10-30:
@tahmidzbr Before i was logging CSI values to csi.dat file without a problem. Then today when i tried, i was getting the same problem as yours. To solve this problem, i just reset my router and started the logging again by running the same commands. Maybe you could try this out.