dhalperi/linux-80211n-csitool-supplementary #240: The sample rate

chendachong, 2016-09-07:
I use a router and a desktop with IWL5300 to get csi information. Now,there is only about 1 sample per second, what can I set to get more samples such as 300 samples per second.

tahmidzbr, 2016-10-01:
@chendachong did you mean you get like 1 packet with 64 bytes per second?

ccjack, 2016-10-19:
If you are talking about how to increase the ping per second to get more CSI, you could use sudo ping -n -i 0.003 to reach your gaol.

tahmidzbr, 2016-10-19:
@ccjack i just figured it out before an hour!
for anyone else if you also want to control the number of packets use this
sudo ping -i 0.05 (IP) -c (100)
100=number of packets!

tahmidzbr, 2016-10-19:
@ccjack hi there, can you capture the csi? I cant seem to log_to_file csi.dat nothing gets recorded in my csi.dat file. Can you share what you did?

ccjack, 2016-10-19:
Yes, I got like 100Hz rate csi data in client mode (Linux computer with 5300 as cient while connecting to the normal AP). Using iw dev wlan0 link to check whether connection exists

tahmidzbr, 2016-10-19:
@ccjack 
My connection with AP is fine. I can transmit and get replies when I ping from another terminal. The problem is when i log_to_file csi.dat i see nothing is getting recorded. I know my AP is transmitting at HT rate. The way i know is my running the rate_scale_table i can see the last Tx was at HT rate. So i am sure i get n rates.

But if i log_to_file csi.dat and then ping from another terminal, even after 1 hour i still dont get any CSI.

Can you please share how you got the CSI? Please your help is greatly appreciated!

ccjack, 2016-10-19:
I follow all the install instruction in the website. After that, I do ifconfing wlan0 192.168.1.10 and iw dev wlan0 connect "xxx(ssid)". And then, I run log_to_file while pinging 192.168.1.1. Then, it start to record the csi data. Also, try other routers. I noticed that some routers ( linksys wifi router) don't work well. The NETGEAR routes work well for me, at least it rarely break the connection. You can turn your smartphone into AP (wifi hotspot) and repeat the same process again.  Also try other intel wifi card, one of my intel wifi card is so easy to be crashed.

tahmidzbr, 2016-10-19:
@ccjack thank you. I will try some other routers and post an update here.
Also im assuming you started this installation recently, would you like to collaborate with me and measure the amplitude and phase of the CSI separately? there are some papers which say how to calibrate the phase, for me both amplitude and phase needs to be explicitly measured. If you wish, we two can follow the algorithms and get this 2 measurements. After that we go our ways to do our own research (i.e. use these 2 features to do other work)?
if you're interested please email me at tahmidzbr@gmail.com.
(let me know even if ur not interested :) )

ogutiann, 2016-10-30:
@tahmidzbr Hi. Please could you help me on how to set the samples to be recorded at 100 packets per second?  just to clarify, does this mean the same thing as saying the sampling frequency is 100Hz?

ogutiann, 2016-10-30:
@tahmidzbr I am getting the same problem that you have now. before, i could ping and collect CSI values. now when i ping and log to file, the CSI file is empty. 0 bytes!! what is happening???

ogutiann, 2016-10-30:
@ccjack hello Jack, could you please share the command of how to set the sampling rate to 100Hz? i am also using the client mode for collecting the CSI values. I am stuck at this point and i am not sure how to go about doing this. Thank you.

tahmidzbr, 2016-10-30:
@ogutiann you can ping with the following command
sudo ping -i 0.01 IP 
the 0.05 means a packet sent every 0.01s  you can adjust this to waht you need
the sudo b/c i think the min time to send a packet is set by 200ms, so using sudo helps.

regarding sampling rate im not too sure. See the thing is when i ping i see 64 bytes per packet sent and received...but in the terminal where CSI is measured i see i receive 393 bytes. So im not sure why the 393. Do you know why?

also 64 bytes per packet and say we ping 100 packets per second, so it means total of 6.4kbytes per second. how does this relate to the sampling rate? If you knw the answer to these 2 questions pls help!

also we can collaborate in detail specially if ur interested to achieve phase as well as amplitude. my id is tahmidzbr@ece.ubc.ca

tahmidzbr, 2016-10-30:
sorry type: i meant 0.01

tahmidzbr, 2016-10-31:
@ogutiann do you know how to set the transmit antenna to only 1? In my case the different rates that I try to echo (given by my router) all end up with either a 2x3x30 or 3x3x30 CSI matrix. B/c i believe the squeeze (csi) function will not work unless its a 1x3x30 matrix. can you please advise?

swashah, 2016-11-20:
@tahmidzbr : Are you using commercial AP or its setup on laptop with INTEL 5300 NIC?

tahmidzbr, 2017-01-07:
@swashah Im using commercial AP

nafiul-nipu, 2017-07-20:
@tahmidzbr  hello.. can you tell me how you've overcome the problem of 2x3x30 matrix and got 1x3x30 matrix.
and can you tell me how exactly you got 30 sub-carriers. in my case i get a lot more like 2x3x2669???