dhalperi/linux-80211n-csitool-supplementary #241: ping not working with my AP

tahmidzbr, 2016-09-13:
Hi Daniel, Firstly thanks for making this fantastic piece of software for us to do our research. Its allowing us to think directions that we would never think if we didnt have this tool that you provided.

However, after following the installations instructions very carefully, I am right now stuck in the final step. The part where I need to ping to my IP address (AP) to get the CSI measurements. The problem is everytime I try to ping to the ip address of the AP (which is a TP-LINK router without any internet connection) it keeps saying "Network unreachable". But i also tried pinging to 0.0.0.0 and then i see that some measurements are made but what i dont understand is why is it 0.0.0.0 as opposed to my actual IP? Also when i run the command route -n i can see the IP address of 163.xxx.xx.xxx which is little weird b/c i know my IP address is 192.168.xxx.xxx. I also tried pinging it to the 163..xxx.xxx.xxx IP address and when i do, i get a message "Do you want to ping broadcast?" 

Can you please advise me what im doing wrong, and how I can configure it so that I can ping to my correct IP address?

Many thanks. 

Tahmid

niuqun, 2016-09-23:
@hmlsi reported similar problems in his thread (https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/45) without showing details. Please let us know if you have resolved this problem. Thank you!

tahmidzbr, 2016-09-23:
@niuqun  I couldnt yet solve the problem, hoping somebody may help us.

niuqun, 2016-09-24:
@tahmidzbr  I think you should try this (https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/234). You need to bring up eth0 manually according to @zhangshaohu so that you can get the right IP address for your client. But I do not know the reason for this.

tahmidzbr, 2016-10-01:
@niuqun  I went over his code, and it seems to be okay now. Is it working for you or are you okay?
Also i wanted to know what is the packet duration and the interval between packets sent? L
Packet duration: i.e. how long does a packet last before the next packet arrives?
Interval: can we adjust the packet timing? like say send one packet and then wait 2 seconds before sending another? or is it fixed?

I know every packet i receive is 64 bytes. What does this 64 bytes contain? Any symbol? is it fixed?
What do you think mate?

tahmidzbr, 2016-10-14:
@niuqun Hi would you like to collaborate to verify that the data we get is similar? I.e. the data we collect in terms of phase and amplitude from CSI matrices? I know it wont be same, but atleast if we know that our results show the same relative values so its correct? I think it will be beneficial in our research process atleast to the point that our data collection from WIFI access points are valid.