dhalperi/linux-80211n-csitool-supplementary #242: How to get the CSIs from 128 subcarriers ？

mikeWangJC, 2016-09-24:
Hi guys
Is there anyone who tried to get the CSIs from all the 128 subcarriers?
If someone has done this before ,plz let me know if you successed ?and if it's possible plz give some 
advice about how to get them .
thanks so much !

brunosoaresds, 2016-11-15:
@mikeWangJC The intel 5300 NIC group the subcarriers in 30 "virtual" subcarriers in the CSI trace. So if you are using 20 or 40MHz you will only get 30 subcarriers in the CSI trace, however each CSI subcarrier will be a group of some X indexed subcarriers. All of this is inside 802.11n pattern, read the CSI chapter and you will understand better :)

mikeWangJC, 2016-11-15:
@brunosoaresds dude thank you so much